MarlinFirmware/Marlin #20798: [FR] handheld extruder/3d pen control

drunkscientist, 2021-01-16:
### Description

hand held extruder- like a 3d pen, but less portable. i already have a mockup from a bolt/pc heatsink, and im working on modeling a handle to hold an e3dV6 or similar, but as marlin is so massively vast ill just describe what i want to do with my hardware: started with an anet a8 (prusa i3 clone), ramps1.4, one extruder, heated bed, no cooling fan. i want to set the board up for basically a second extruder, bowden fed, which i will hold and will have a trigger for extrusion, ideally with some degree of speed control. basically a wired, desktop 3d pen.

### Feature Workflow

probably a menu entry to set mode, which would prompt for a temperature, then wait for the appropriate heater to reach said temp, before displaying 'ready' and maybe the total distance extruded or something. the user presses on a trigger to extrude filament similar to a hot glue gun or wire based welding process

#### Additional Information

ive been trying to do this on my own for kindof a while. i had to move and packed it away for a bit and recently came back to find that marlin had gotten even bigger and more complicated. i know all the parts are here i just dont know where the relevant bits are or enough about programming to know how to implement a new mode. but i am trying to learn, so even just pointers for where to look would be appreciated

PomponXZ, 2021-01-18:
to be honest, isnt it easier and cheaper to just wire together a 555 with variable speed and a microswitch to turn it on and off, to a stepper motor driver (555 output to step pin) and use that instead of an arduino? you could just use an attiny with the arduino bootloader (or an arduino nano) and a basic PID program to control the heater and you can grab the calibration table from marlin so no need to calibrate either.  (you can use a button and some leds as a display to select the temperature.)

Programing all of this into marlin for a use case which no one would use, and that to be fair, what i told you is a far better solution (you could even make it into an arduino uno/nano shield and it would be neat). Also you are far better programing it from scratch  than trying to modify marlin. You can even do away with the 555 and use the arduino to generate  the signal.
So all in all , all you need is an arduino, a mosfet, a potenciometer, and a few butons and resistors and leds. the rest is software.

A more sensible approach would be to add "external motion controls" like some CNC milling machines have, either buttons or encoders as inputs. I have no idea, but maybe grbl which is more cnc oriented already has that.

drunkscientist, 2021-01-19:
so someone already made a patch for joystick control/jogging, which [was apparently added at some point](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/Configuration_adv.h#L3358), how is that more useful that what im asking? 'add external controls' is i guess an appropriate heading, but i only need one stepper and one heater, how is cnc software (which to my knowledge does not account for heaters) more relevant?

and yes i do plan to eventually boil it down to a nano, a mosfet, and a stepper driver, the rest IS software, but the bar for learning everything required is a lot higher than the bar to compile/upload a new version of marlin, which literally has all the software components, plus saftey features