MarlinFirmware/Marlin #20803: Feature Request, Distinct Z Factors

jasonfox7, 2021-01-17:
Is there the possibility of adding distinct Z factors in the future?

I have 3 z steppers and one of the leadscrews is slightly out of spec with the others resulting in a .5mm drop per 100mm of travel. It's driving me mad. A distinct steps per mm would fix it all.

Thanks!

sjasonsmith, 2021-01-17:
I think the idea has merit, but it isn't viable in today's Marlin architecture.

Multiple drivers on one axis is achieved today by simply sending step pulses to multiple pins. It is strongly assumed that they always operate at the same step rate.

A pretty significant rework to Marlin's motion mechanisms would be needed to allow for this. Those changes might be desirable, but would be very large in scope. It is going to be much easier to just replace that defective screw.

jasonfox7, 2021-01-18:
Fair enough. :)

closed 2021-01-18 by jasonfox7