MarlinFirmware/Marlin #20805: [BUG] (short description)

BrianP310, 2021-01-18:
### Bug Description

I have an Anycubic Chiron which I am upgrading to an SKR 1.4 Turbo and a combination of 2130 and 2208 drivers.  I want to use the original screen on the Chiron.  I have Marlin compiling all the way to "Linking...firmware.elf", then ld.exe gives me a bunch of errors about "undefined reference to 'Anycubic::Chiron'".  I cannot for the life of me figure out what it's talking about, since all reference appear to be in the header files, as needed, and the respective .o files are in the folder it specifies.  
I have never used Marlin (always used Klipper on my other machine), but I would very much like to get this working with the original screen and the SKR.  Any help would be appreciated.  
I am using Visual Studio Code with PlatformIO (which I've never used before, either) so it is possible that I have an environment issue, though I don't believe that's it.  
Thanks in advance!

### Configuration Files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5827333/Marlin.zip)
[platformio.zip](https://github.com/MarlinFirmware/Marlin/files/5827350/platformio.zip)

I had to add the following pin definition to pins_BTT_SKR_V1_4.h:
#ifndef Z_MAX_PIN
     #define Z_MAX_PIN                        P1_00  // PWRDET
#endif

and the following pin definition to pins_BTT_SKR_V1_4_TURBO.h (no reason I did it in two separate files, just worked out that way.)
#ifndef BEEPER_PIN
    #define BEEPER_PIN P1_30
#endif

### Steps to Reproduce

1. Make pin definitions listed above
2. Copy configuration[_adv].h files
3. Compile with PlatformIO

**Expected behavior:**

Successful Compile

**Actual behavior:**

Everything compiles fine until "Linking .pio\build\LPC1769\firmware.elf" when I get the following errors and fails:

c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o: in function `ExtUI::onStartup()':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:40: undefined reference to `Anycubic::Chiron'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o: in function `ExtUI::onIdle()':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:42: undefined reference to `Anycubic::Chiron'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o: in function `ExtUI::onPrinterKilled(char const*, char const*)':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:45: undefined reference to `Anycubic::Chiron'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o: in function `ExtUI::onMediaInserted()':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:48: undefined reference to `Anycubic::Chiron'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o: in function `ExtUI::onMediaError()':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:49: undefined reference to `Anycubic::Chiron'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/anycubic_chiron_lcd.cpp.o:C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui/anycubic_chiron_lcd.cpp:50: more undefined references to `Anycubic::Chiron' follow
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/lib/anycubic_chiron/chiron_tft.cpp.o: in function `Anycubic::ChironTFT::SendFileList(signed char)':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui\lib\anycubic_chiron/chiron_tft.cpp:398: undefined reference to `Anycubic::filenavigator'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/lib/anycubic_chiron/chiron_tft.cpp.o: in function `Anycubic::ChironTFT::SelectFile()':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui\lib\anycubic_chiron/chiron_tft.cpp:420: undefined reference to `Anycubic::filenavigator'
c:/users/brian/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/LPC1769/src/src/lcd/extui/lib/anycubic_chiron/chiron_tft.cpp.o: in function `Anycubic::ChironTFT::PanelAction(unsigned char)':
C:\Users\brian\Downloads\Marlin-bugfix-2.0.x (1)\Marlin-bugfix-2.0.x/Marlin\src\lcd\extui\lib\anycubic_chiron/chiron_tft.cpp:692: undefined reference to `Anycubic::filenavigator'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\LPC1769\firmware.elf] Error 1
================================================================================ [FAILED] Took 167.26 seconds ================================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:02:47.261
=========================================================================== 1 failed, 0 succeeded in 00:02:47.261 =========================================================================== 
The terminal process "C:\Users\brian\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.

#### Additional Information

[image]

sjasonsmith, 2021-01-18:
I has to fix several issues to get this to compile. Please test the changes in the above PR (#20807) to see if they work for you.

I don't have the printer or the display, so don't have any way to test it beyond compilation.

BrianP310, 2021-01-18:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5831002/Configuration.zip)

@sjasonsmith Thanks, that definitely allows it to compile now.  I had to change my Configuration.h because the TFT connector on the SKR 1.4T is Serial 0, so has to set 

#define LCD_SERIAL_PORT 0.  

This makes the following changes necessary:

#define SERIAL_PORT 1
#define SERIAL_PORT_2 -1

Also, annoyingly, the stock Chiron LCD has the TFT connector wired in a non-standard way.  So to make it work with the SKR 1.4T, you have to swap the pins around as follows (Luckily, if your colors are different, the pins definitions are silk screened on the back of the LCD circuit board, so just match up the colors, as necessary.)

|-------------------------|
|     G    Bk   Y   R   Bl     |        G=RST, Bk=Tx (Goes to Rx on SKR), Y=Rx (Goes to Tx), R=Vss (Gnd), Bl=5V
|------|----------|--------|

So, to summarize, in total to fix this bug, we need to merge your code changes, add the two pin definitions:

#ifndef BEEPER_PIN
    #define BEEPER_PIN P1_30
#endif

//
// Second Z-Stop Pin
//
#ifndef Z_MAX_PIN
  #define Z_MAX_PIN                        P1_00  // PWRDET
#endif

and rewire the TFT connector.  Seems this was definitely a bug, but it is working now.
[image]

BrianP310, 2021-01-18:
@sjasonsmith Do I need to do anything to close this or tag it or anything, or will it close automatically after it's been added?  Sorry for the newb question, but this is my first bug report.

sjasonsmith, 2021-01-18:
I would wait for the change to be merged, then you can close it yourself if it still seems good.

closed 2021-01-24 by sjasonsmith

Piotrekk91S, 2021-02-21:
hello, can you provide marlin for stock lcd?

Piotrekk91S, 2021-02-21:
witam, czy mozesz udostepnic marlina dla oryginalnego wyswietlacza?

Piotrekk91S, 2021-02-21:
hello hello, can you provide the marlin for the original display