MarlinFirmware/Marlin #20809: [BUG][Ender 3 V2] - Disappearing status menu while printing

steljwagh, 2021-01-18:
Sometimes (I did not find a way yet to reproduce), the status screen is disappearing completely trying to get into the Tune menu.

The print will continue, but usually I'll stop the print because of I'd like to see the status, I have no other management options other then the screen of the Creality Ender 3 v2.

Steps to reproduce
1. Start print (doesn't matter how far the print is)
2. Rotate encoder to Tune menu
3. Press button to enter (you'll here the beep, but nothing happens)
4. Press button again to enter the menu

**Expected behavior:**

Get into the tune menu

**Actual behavior:**

Menu returns to Main screen

#### Additional Information

Finally I recognized the BUG at step 3, that's when I grabbed my camera to film the event. So you can hear it coming when you only hear the beep at step 3.

steljwagh, 2021-01-18:
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5830026/config.zip)

https://user-images.githubusercontent.com/20403141/104912666-d7751480-598c-11eb-8c90-7d96f5a095e8.mp4

cloudmonsterrepo, 2021-01-18:
I'm experiencing the same error just like when restarting an impression the previous progress appears if it's the same gcode

steljwagh, 2021-01-18:
> 
> 
> I'm experiencing the same error just like when restarting an impression the previous progress appears if it's the same gcode

Not exactly the same but I do know what you are talking about, if you finished 1 print called A.gcode and after that you'll start B.gcode, it displays A.gcode in screen with some weird glitched progress.

I think that's another bug.

KaraBrandsen, 2021-02-08:
I experienced the exact same bug. I found that this only happens on the 2nd+ print. If you reset the printer after each print you never experience this bug. It probably has something todo with a state not properly clearing after a print completes.

closed 2021-03-21 by github-actions[bot]