MarlinFirmware/Marlin #20815: [BUG] Nozzle temp didn't go back to M109 temp after filament runout and HOTEND_IDLE_TIMEOUT

m-kozlowski, 2021-01-19:
### Bug Description

Got filament runout during print. It was properly detected, nozzle was parked, but as I haven't reacted fast enough, the nozzle was cooled down according to HOTEND_IDLE_TIMEOUT configuration.
I got standard "pres knob to heat up" message. Then nozzle was heat up to 170 degree (EXTRUDE_MINTEMP) and prompt to insert filament was displayed. After inserting filament and confirming that no more purges are needed, printing resumed at 170C.
Before runout the print was running at 235C, which I think should be restored at filament change.

I'm running bugfix-2.0.x e8aa6ab7354ee438067c605b72907d9bf69df3fb (Jan 12)

**Expected behavior:**
Restore printing temp. set with M109/M104.

**Actual behavior:**
Nozzle was preheat to EXTRUDE_MINTEMP and resumed print whithout restoring required temperature

slowbro, 2021-01-20:
I just tried to replicate this, and was not able to. After runout, I let the printer sit for more than the HOTEND_IDLE_TIMEOUT, and after re-loading the filament, it resumed at the M109 temp (195) instead of the EXTRUDE_MINTEMP (170).

 Are you able to paste your config, and possibly the gcode you were printing?

m-kozlowski, 2021-01-20:
Sure.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5842359/Configuration.zip)
[mijia_case_v1_0.6n_0.3mm_PETG__7h23m.zip](https://github.com/MarlinFirmware/Marlin/files/5842362/mijia_case_v1_0.6n_0.3mm_PETG__7h23m.zip)

m-kozlowski, 2021-01-20:
So:
- timeout disabled hotend ad bed as well
- reheat heated nozzle only to 170C (which is too low for petg anyway)
- "continue" after inserting filament did not change temperature at all (also keeping bed set at 0C)

slowbro, 2021-01-20:
Thanks, I'll take a look and try to replicate it again tonight. Were you printing from serial?

m-kozlowski, 2021-01-20:
Yes, both cases. However reheat, filament change and resume was handled with lcd.

slowbro, 2021-01-21:
Hmm, okay, I think I see the problem, and why I wasn't able to reproduce it with my config. The tl;dr to fix your problem is to lower the `PAUSE_PARK_NOZZLE_TIMEOUT` setting to a value below the `HOTEND_IDLE_TIMEOUT_SEC` setting, so that the pause feature's timeout happens first.

---

But, the underlying bug should still be fixed. There are two competing settings here:

1. PAUSE_PARK_NOZZLE_TIMEOUT
2. HOTEND_IDLE_TIMEOUT_SEC

In my case, I have `HOTEND_IDLE_TIMEOUT_SEC` at 5min, and `PAUSE_PARK_NOZZLE_TIMEOUT` at 45s - the defaults. This means that the `ADVANCED_PAUSE` feature will cool the hotend before the `HOTEND_IDLE_TIMEOUT` feature can. However, in your case, you have these set to 5 min & 10 min respectively. So, the hotend idle timeout kicks in & drops the temp to 0, and this part of `pause.cpp` sets it to `EXTRUDE_MIN_TEMP` when you go to reload:

https://github.com/MarlinFirmware/Marlin/blob/5ee1087959f88dc60386ff3caa21e75d9e20b128/Marlin/src/feature/pause.cpp#L131-L160

There are a number of ways I can see to fix this.

1. Disable the hotend idle timeout in `pause.cpp` if PAUSE_PARK_NOZZLE_TIMEOUT is set
2. Have the `thermalManager` (`Temperature` class) keep track of the previous temperature so we can recall it later
3. Have the pause routine(s) keep track of the previous temperature so we can recall it later
4. Have the hotend timeout routine(s)... " "
5. Make HOTEND_IDLE_TIMEOUT and PAUSE_PARK_NOZZLE_TIMEOUT mutually exclusive

`1` makes the most sense to me, but I would worry about the safety implications of diabling the hotend idle timeout. `2-4` seem okay, but it will cost memory to keep the value around. Of all the places to store it, I feel like `3` makes the most sense unless there is some explicit need to keep the value around for other features. The stored value can then be used in place of `extrude_min_temp` in the above-referenced function. `5` seems like the nuclear option.

m-kozlowski, 2021-01-21:
After decreasing PAUSE_PARK_NOZZLE_TIMEOUT nozzle reheats to printing temperature and bed stays on, so everything works as expected. Thank you :)

My 2 cents regarding final solution, if anyone cares: 
I'd vote for (1). There shouldn't be safety implications to worry about as long as (reasonable) PAUSE_PARK_NOZZLE_TIMEOUT is defined - nozzle will be turned down one way or another.
However generic push/pop mechanism for thermalManager _might_ be more future-proof. 
(5) Naah. Both mechanisms covers different use cases and selecting one shouldn't rule out the other.
But it might be good idea to add a sanity check that PAUSE_PARK_NOZZLE_TIMEOUT is less or equal to HOTEND_IDLE_TIMEOUT

slowbro, 2021-01-21:
Awesome! Glad that fixed it.

Thanks for the input. I'm curious if @thinkyhead has any input on the options; I can work on whichever is deemed the best path forward.

slowbro, 2021-01-24:
@sjasonsmith I'm curious if you have an opinion on the best fix for this issue. Itching to write some code :)

sjasonsmith, 2021-01-24:
@slowbro I do not have any opinions on this. I never use the pause functionality, so I am unfamiliar with the workflows it involves.

sjasonsmith, 2021-01-24:
Your best bet is to just post what you think is the best approach as a pull request.

slowbro, 2021-01-25:
Alright - I'll work on a solution. Thanks!

closed 2021-03-03 by github-actions[bot]