MarlinFirmware/Marlin #20818: PROBE_OFFSET_WIZARD - Crashing printer

galluccib, 2021-01-19:
Current branch and when going to PROBE Z Wizard it says homing and the system just reboots. Went back to make sure I had it in the conf adv correctly and noticed was getting a build error on "probe_offset_wizard"

Config -
 // Add Probe Z Offset calibration to the Z Probe Offsets menu
  #if HAS_BED_PROBE
    #define PROBE_OFFSET_WIZARD
    #if ENABLED(PROBE_OFFSET_WIZARD)
      #define PROBE_OFFSET_START -4.0   // Estimated nozzle-to-probe Z offset, plus a little extra
    #endif
  #endif

Build Error - 

.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `probe_offset_wizard_menu()':
C:\Temp\Compile\New\Marlin-2.0.x\Marlin-2.0.x/Marlin\src\lcd\menu/menu_probe_offset.cpp:93: warning: undefined reference to `lcd_moveto(unsigned char, unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/menu/menu_probe_offset.cpp.o: In function `lcd_put_u8str(char const*)':
C:\Temp\Compile\New\Marlin-2.0.x\Marlin-2.0.x/Marlin\src\lcd\menu\../lcdprint.h:161: warning: undefined reference to `lcd_put_u8str_max(char const*, unsigned short)'

Thanks,

slowbro, 2021-01-20:
Hi @galluccib, can you please post your Configuration.h and Configuration_adv.h? Also, does this also happen when using the `bugfix-2.0.x` branch?

galluccib, 2021-01-20:
I ended up using this and everything worked fine..

https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

slowbro, 2021-01-20:
Gotcha. Looks like that is based on the bugfix branch; but I'm not sure what change exactly fixed that.

slowbro, 2021-01-21:
Looks like it was likely their update back in August that fixed it, I see other references to bedlevelling issues [here](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/60) which was right before they updated to bugfix. So it was either fixed here in bugfix-2.0.x or was fixed by the maintainer of that repo.

rhapsodyv, 2021-01-31:
It was fixed by #20838 and its merged on bugfix.

closed 2021-01-31 by rhapsodyv