MarlinFirmware/Marlin #20821: [FR] Rotate graphical LCD orientation (upside-down)

thiccnugget, 2021-01-19:
### Description

It could be useful to some users (me too) to have a feature in the Marlin firmware to rotate the LCD upside-down to fit it in custom machines and adapt different LCDs in the same machine.

### Feature Workflow

It would be nice to have something like:

`#define LCD_ROTATION 180`  in the Graphical LCDs section

where "180" (degrees) is set to flip the LCD upside-down.
By default, this should be commented or set to 0

#### Additional Information

I remember doing the LCD rotation on Arduino with the u8g library and `u8g.setRot90()` and I thought if this was possible on Marlin.

Since I have this MKS MINI12864 LCD (https://reprap.org/wiki/MKS_MINI_12864), I had to flip it to adapt it to my 3dp printer. So I'm asking if it is possible to implement this function because I think that it may be helpful to many users.

I'm new to Marlin 2.0 and I would like to contribute to it!

ellensp, 2021-01-19:
should already do it

in Marlin/src/lcd/dogm/marlinui_DOGM.cpp

  TERN_(LCD_SCREEN_ROT_90, u8g.setRot90());
  TERN_(LCD_SCREEN_ROT_180, u8g.setRot180());
  TERN_(LCD_SCREEN_ROT_270, u8g.setRot270());

ellensp, 2021-01-19:
Just tred it on a RAMPS
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
#define LCD_SCREEN_ROT_180

works as expected

closed 2021-01-19 by ellensp

thiccnugget, 2021-01-19:
Hi, thank you for the support! How do I edit this code to actually rotate
the screen by 180 degrees? Do I have to comment the first and third lines
(LCD_SCREEN_ROT_90 / 270) ?

ellensp, 2021-01-19:
you add #define LCD_SCREEN_ROT_180 to your Configuration.h file

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2021-01-19:
In some cases it is also in the pins.h file. eg in pins_RAMPS.h

      #if ENABLED(MKS_MINI_12864)

        #define DOGLCD_A0                     27
        #define DOGLCD_CS                     25

        // GLCD features
        // Uncomment screen orientation
        //#define LCD_SCREEN_ROT_90
        //#define LCD_SCREEN_ROT_180
        //#define LCD_SCREEN_ROT_270