MarlinFirmware/Marlin #20824: Flashing Marlin 2.0.7.2 or Jan 18th buxfix release on Malyan M200 v2 results in boot loop

doucettom, 2021-01-19:
# NO SUPPORT REQUESTS PLEASE

Do you want to ask a question? Are you looking for support? Please don't post here. Support Requests posted here will be automatically closed!

Instead use one of the following options:

- The Marlin Firmware forum at https://reprap.org/forum/list.php?415
- The MarlinFirmware Facebook Group at https://www.facebook.com/groups/1049718498464482/
- The MarlinFirmware Discord Server at https://discord.gg/n5NJ59y.

Before filing an issue be sure to test the latest "bugfix" branch to see whether the issue is already addressed.

doucettom, 2021-01-19:
Bugfix files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5837062/Marlin.zip)

thinkyhead, 2021-01-22:
Maybe you can help isolate the cause through testing Marlin from points in time between the version where it works (2.0.6.1) and the version where it stops working (2.0.7). The simplest way to do this is to start with the point in time (T) halfway between the two releases. If you find that the bug is not present, you would test the point in time halfway between T and version 2.0.7. If you find that the bug is present, you would test the point in time halfway between 2.0.6.1 and T.

After only a few tests following this method you should be able to locate the exact point in time where the bug appeared.

doucettom, 2021-02-03:
Alright ! I had some time to do some testing. It seems like this bug appeared somewhere between 2.0.7.1 and 2.0.7.2. 2.0.7.1 is working just fine.

doucettom, 2021-03-06:
I guess I need to add a reply to keep this issue opened. I didn't have much love to give to the Minis lately but I'm still willing to help with testing to solve this

doucettom, 2021-03-19:
Anything I can do to help? I'd like to see the mini's up to date on next release :)

doucettom, 2021-04-30:
Hi ! 2.0.8 just released. Should I test again ?

thisiskeithb, 2021-04-30:
> Hi ! 2.0.8 just released. Should I test again ?

Yes, please.

closed 2021-07-11 by github-actions[bot]