MarlinFirmware/Marlin #20825: Z Axis makes noise with TMC2209

saintofinternet, 2021-01-19:
Hi guys I'm a complete newbie here but I compiled the latest bugfix last night, I was previously running the pre-compiled BTT firmware.

Seemed good, was storing my settings, set a print going and the Z is insanely loud I thought something was wrong I ran upstairs to check haha!

Commenting out HYBRID_THRESHOLD fixed it for me though.

_Originally posted by @ashkwil in https://github.com/MarlinFirmware/Marlin/issues/16407#issuecomment-618906465_

in my Marlin Bugfix 2.0.x the HYBRID_THRESHOLD is already disabled...

still there is noise in the Z motors

swilkens, 2021-01-19:
Please follow the issue report template and add you configuration files, also please tell us what your printer is / which hardware.

I'm going to assume this is an Ender 3 and you replaced the standard board with an SKR mini E3?

Loud Z-axis could be that it's stuck in spreadcycle mode, acceleration could be too high, it could be missing steps. Difficult to say without further information :)

saintofinternet, 2021-01-20:
hi,
it's a Hypercube Evolution.

also can you please guide me where the issue report template is?

thank you for your reply!

ellensp, 2021-01-20:
Stock reply, with a copy of the template

```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

saintofinternet, 2021-01-20:
Thank you very much for the guide!

closed 2021-02-26 by github-actions[bot]