MarlinFirmware/Marlin #20827: [BUG] Warnings emitted when UBL_DEVEL_DEBUGGING enabled

slowbro, 2021-01-19:
### Bug Description

When enabling UBL_DEVEL_DEBUGGING (to get the `G29 W` command), the build has warnings:

```
In file included from Marlin/src/feature/bedlevel/ubl/../../../inc/MarlinConfig.h:55:0,
                 from Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:23:        
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp: In static member function 'static void unified_bed_leveling::g29_what_command()':
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1707:93: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
         SERIAL_ECHOLNPAIR("Meshes go from ", hex_address((void*)settings.meshes_start_index()), " to ", hex_address((void*)settings.meshes_end_index()));
                                                                                             ^
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:116:57: note: in definition of macro '_SEP_2'
 #define _SEP_2(PRE,V)     serial_echopair_PGM(PSTR(PRE),V)                 
                                                         ^                                                             
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:173:28: note: in expansion of macro '_SELP_4'
 #define __SELP_N(N,V...)   _SELP_##N(V)                                                                               
                            ^~~~~~                                                                                     
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:174:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)                                                                              
                            ^~~~~~~~                                                                                   
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:206:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)                                                                
                                 ^~~~~~~                                                                               
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1707:9: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
         SERIAL_ECHOLNPAIR("Meshes go from ", hex_address((void*)settings.meshes_start_index()), " to ", hex_address((void*)settings.meshes_end_index()));
         ^~~~~~~~~~~~~~~~~                                                                                             
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1707:150: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
         SERIAL_ECHOLNPAIR("Meshes go from ", hex_address((void*)settings.meshes_start_index()), " to ", hex_address((void*)settings.meshes_end_index()));
                                                                                                                                                      ^
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:176:62: note: in definition of macro '_SELP_2'
 #define _SELP_2(PRE,V)     do{ serial_echopair_PGM(PSTR(PRE),V); SERIAL_EOL(); }while(0)
                                                              ^          
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:173:28: note: in expansion of macro '_SELP_4'
 #define __SELP_N(N,V...)   _SELP_##N(V)                                                                               
                            ^~~~~~                                                                                     
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:174:28: note: in expansion of macro '__SELP_N'
 #define _SELP_N(N,V...)    __SELP_N(N,V)                                                                              
                            ^~~~~~~~                                                                                   
Marlin/src/feature/bedlevel/ubl/../../../inc/../core/serial.h:206:33: note: in expansion of macro '_SELP_N'
 #define SERIAL_ECHOLNPAIR(V...) _SELP_N(NUM_ARGS(V),V)                                                                
                                 ^~~~~~~                                                                               
Marlin/src/feature/bedlevel/ubl/ubl_G29.cpp:1707:9: note: in expansion of macro 'SERIAL_ECHOLNPAIR'
         SERIAL_ECHOLNPAIR("Meshes go from ", hex_address((void*)settings.meshes_start_index()), " to ", hex_address((void*)settings.meshes_end_index()));
         ^~~~~~~~~~~~~~~~~
```

They don't fail the build (because they are warnings).. I'm not sure how important this is, overall, but thought I would at least bring it up.

### Configuration Files

Attached: [config.zip](https://github.com/MarlinFirmware/Marlin/files/5838657/config.zip)

I added `#define UBL_DEVEL_DEBUGGING` to the ubl section, though it appears to be usually defined in `src/feature/bedlevel/ubl/ubl.h`

### Steps to Reproduce

1. Enable devel debugging w/ `#define UBL_DEVEL_DEBUGGING`
2. Build Marlin.

**Expected behavior:**

Warnings would not be emitted.

**Actual behavior:**

Warnings are emitted.

sjasonsmith, 2021-01-24:
This is a development feature intended to be used by developers, which is why it is not in Configuration.h. If you are actively encountering these, please fix the warnings and post a pull request. I don't think there is value in keeping issues open for warnings which are not normally visible to end users.

closed 2021-01-24 by sjasonsmith

slowbro, 2021-01-24:
Fair enough; I'll see what I can do to fix it. I was afraid to mess with something that seemed super low-level, but I'll give it a shot because how else does one learn things?