MarlinFirmware/Marlin #20833: [BUG] "LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."

G3rb, 2021-01-20:
### Bug Description

I can't compile the last bugfix version because of this error.

"LCD_SERIAL_PORT cannot be the same as SERIAL_PORT_2."

I have a creality ender 3v2 and I used the configuration exemple you provide (the bugfix branch version) I have added the bltouch and runnout sensor so I set 

#define SERIAL_PORT_2 3

in Configuration.h but the SanityCheck script break the compilation and print this error.

The compilation works fine with this parameter on the default branch.

### Configuration Files

the one you provide on the bugfix branch for the ender 3v2 with the

#define SERIAL_PORT_2 3

option set (+ bltouch and runnout sensor options set)

ellensp, 2021-01-20:
Not a bug.
DO NOT #define SERIAL_PORT_2 3
Marlin automatically sets up  LCD_SERIAL_PORT and port 3 for this machine.

ellensp, 2021-01-20:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-20 by ellensp

G3rb, 2021-01-20:
Ok, sorry.