MarlinFirmware/Marlin #20834: G35 AND G29 SHOW DIFFERNT VALUES.

Talha909, 2021-01-20:
I have printer with SKR 1.4 Tubo and TMC 2130 . I have two z drivers on Z.  I did the G29 and G35.  on 0 point my value is -0.163 and last point +0.114. When I did the G35 then the values are different. I did this 5 times or more times. Result was always different.  Front left is in negative value and front right is in positive. And when I more the front right CW 1 minutes after that's its values increase.  I am using the BUG FIXED DATED 8-1-2021

Bilinear Leveling Grid:
      0      1      2
 0 -0.163 +0.010 +0.114
 1 +0.124 -0.024 +0.017
 2 +0.113 +0.018 +0.145
Turn Front-Right CW by 0 turns and 1 minutes (-0.02mm)
Turn Back-Right CCW by 0 turns and 1 minutes (0.01mm)
Turn Back-Left CW by 0 turns and 1 minutes (-0.01mm)
G28
G29
G35
Bilinear Leveling Grid:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5842613/Marlin.zip)

      0      1      2
 0 -0.157 +0.029 +0.124
 1 +0.146 -0.007 +0.041
 2 +0.154 +0.045 +0.159
Turn Front-Right CW by 0 turns and 2 minutes (-0.03mm)
Turn Back-Right CW by 0 turns and 1 minutes (-0.01mm)
Turn Back-Left CCW by 0 turns (0.01mm)

ellensp, 2021-01-21:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sjasonsmith, 2021-01-24:
Closing, since the template was not followed and it is not clear what the error is.

closed 2021-01-24 by sjasonsmith

SIWONASSI, 2021-01-28:
Sorry friend, I know it is not the place, but you managed to solve this problem, X Lite Pro calibration problem after factory reset, all over the internet you only mentioned this problem, if you had any solution you can tell me. EMAIL: siwonassi@hotmail.com