MarlinFirmware/Marlin #20839: [BUG] (short description) Error while compiling MAX31865 with PT100 sensor

fox631, 2021-01-21:
Когда я компилирую прошивку с программным SPI, для датчика ПТ100 с платой MAX31865, я получаю ошибку:
Marlin\src\module../libs/softspi.h:85:70: error: cannot convert 'bool' to 'WiringPinMode' for argument '2' to 'void pinMode(uint8, WiringPinMode)'

He does not like the file Marlin \ src \ libs \ softspi.h, this line:
inline void fastPinMode (uint8_t pin, bool mode) {pinMode (pin, mode); }

ellensp, 2021-01-22:
```
Please follow the instructions below. Failure to do so may result in your issue being closed.

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

<THE TEMPLATE>

### Bug Description

<!-- Describe the bug in this section. (You can remove this invisible comment.) -->

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

<!-- Describe the steps needed to reproduce the issue. (You can remove this invisible comment.) -->

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

<!-- Describe what you expected to happen here. (You can remove this invisible comment.) -->

**Actual behavior:**

<!-- Describe what actually happens here. (You can remove this invisible comment.) -->

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

fox631, 2021-01-22:
### Bug Description

When I compile the firmware with software SPI, for a PT100 sensor with a MAX31865 board, I get the error:
Marlin\src\module../libs/softspi.h:85:70: error: cannot convert 'bool' to 'WiringPinMode' for argument '2' to 'void pinMode(uint8, WiringPinMode)'

He does not like the file Marlin \ src \ libs \ softspi.h, this line:
inline void fastPinMode (uint8_t pin, bool mode) {pinMode (pin, mode); }

The bug is repeated in Marlin versions 2.0.7.1, 2.0.6.1 and Marlin-bugfix-2.0.x

I am using SKR mini E3 V1.2 board and PlatformIO compilation environment

### Configuration Files

[files.zip](https://github.com/MarlinFirmware/Marlin/files/5856755/files.zip)
I wrote to file platformio.ini:
default_envs = STM32F103RC_btt

To file Marlin\src\pins\stm32f1\pins_BTT_SKR_MINI_E3_common.h:
#define TEMP_0_PIN                          PA9   // Analog Input "TH0"
#define MAX31865_CS_PIN                     TEMP_0_PIN
#define MAX31865_MOSI_PIN                   PB9
#define MAX6675_DO_PIN                      PB5
#define MAX6675_SCK_PIN                     PB6
#define MAX6675_SS_PIN                      PB8

To file Marlin\Configuration.h
#define MAX31865_SENSOR_OHMS      100
#define MAX31865_CALIBRATION_OHMS 430
#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
#define SERIAL_PORT -1
#define SERIAL_PORT_2 2
#define BAUDRATE 115200
#define TEMP_SENSOR_0 -5

To file Marlin\Configuration_adv.h:
#define MONITOR_DRIVER_STATUS
#define TMC_DEBUG
#define SHOW_TEMP_ADC_VALUES
#define MAX_CONSECUTIVE_LOW_TEMPERATURE_ERROR_ALLOWED 10

### Steps to Reproduce

1. Insert my files into Marlin firmware
2. Start compilation

**Expected behavior:**

The firmware should have compiled
**Actual behavior:**

During compilation, an error is received:
Marlin\src\module../libs/softspi.h:85:70: error: cannot convert 'bool' to 'WiringPinMode' for argument '2' to 'void pinMode(uint8, WiringPinMode)'

He does not like the file Marlin \ src \ libs \ softspi.h, this line:
inline void fastPinMode (uint8_t pin, bool mode) {pinMode (pin, mode); 

#### Additional Information

I configured the firmware to work with the PT100 and MAX31865 sensor using the information from here:
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/179

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by ellensp

fox631, 2021-02-11:
@GadgetAngel It is written that ellensp closed it yesterday. Do I have to do something else?

GadgetAngel, 2021-02-11:
No, you do not have to do anything else