MarlinFirmware/Marlin #20841: [BUG] Ender 3 V2 no Junction Deviation due to dwin.cpp

hamstertimer, 2021-01-21:
Hi,
I just wanted to switch on my Ender 3 V2 from Jerk to Junction Deviation. After disabling Classic_Jerk in the configuration.h file, I get numerous errors because basically everything in dwin.cpp is only set up for Classic Jerk.  

I'm running the current Marlin 2.0.x.

Malmik, 2021-01-23:
Try the bugfix branch. I am running 4.2.6 board (stock one) on ender 3v2, with junction deviation

sjasonsmith, 2021-01-25:
This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2021-01-25 by sjasonsmith