MarlinFirmware/Marlin #20843: [BUG] Bugfix-Release removed feedrate variable "#define HOMING_FEEDRATE_Z"

Cysign, 2021-01-21:
Buxfix-Release has changes in speed parameters:

configuration.h 1492:
`// Homing speeds (mm/min)
#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (17*60) }`

`Marlin\src\inc/../../Configuration.h:1001:28: error: 'HOMING_FEEDRATE_Z' was not declared in this scope`

Should I just add 
`#define HOMING_FEEDRATE_Z  (17*60)` or will it lead to even more problems? (That's non-bugfix-code from Marlin 2.0.7.2)

Cysign, 2021-01-21:
Adding the old veriable works.
But I believe it's better to update the code ;)

thisiskeithb, 2021-01-21:
You can't mix & match configs with versions of Marlin. If you're running Marlin [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip), use configs from the [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Configurations/archive/bugfix-2.0.x.zip) branch.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-21 by thisiskeithb