MarlinFirmware/Marlin #20844: Auto bed level not working on front right corner only

rodribike, 2021-01-21:
### Bug Description

 Auto bed level not working - front right corner not recognized when probing - pronterface to confirm this - ==== error shows up in this corner 
When printing G28 and G29 commands are performed but when print starts right front corner gets messed up. All other places on bed are ok
BUT if you place on start Gcode 2x G29 probing and printing are ok. 2x G29 means this:
G29
G29

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Send print a part on printer
2. Print on big surface gets messed up on the right front corner
3. Connect printer to pronterface 
4. send G28
5. send G29
6. ===== ERROR shows up on probe readings only on right front side
7. Repeat G29
8. Now all probing points have measurements 
9. Update start g code with G29 x2 
10. G29 
11. G29
12. Slice part and send by SD card to printer 
13. Start print 
14. Print starts.... G28 Is performed G29 is performed, 2nd G29 is performed and now print is ok! Without firmware changes nor manual bed leveling 

**Expected behavior:**

 xpected to print starts normally with compensation all over the bed

**Actual behavior:**

 Right front corner seems to not be compensated

To add that im not the only one with this issue. i know someone that has 3 printers with different boards and probes and this issue happens in all of them! 

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* 
[image]
[image]
[image]
[image]
[image]

[not probing right front corner.zip](https://github.com/MarlinFirmware/Marlin/files/5852466/not.probing.right.front.corner.zip)

* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

daleckystepan, 2021-01-21:
Should be fixed in latest bugfix #20122

ellensp, 2021-01-22:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

rodribike, 2021-01-22:
Hello
I have marlin 2.0.7.2 
I believe it's the version you mention no?

ellensp, 2021-01-22:
not that is release 2.0.7.2 not bugfix

sjasonsmith, 2021-01-24:
This is a known issue which is already fixed in `bugfix-2.0.x`. Please use that or disable `BILINEAR_SUBDIVISION` until 2.0.8 is released.

closed 2021-01-24 by sjasonsmith