MarlinFirmware/Marlin #20845: [BUG] IWDG does not work properly.

mixware011, 2021-01-22:
Hi,
i was using BOARD_MKS_ROBIN_NANO_V2.
I was using the latest firmware, but this motherboard crashed occasionally, and the nozzle kept heating after the crash, causing my thermistor damage.
I checked the source code and found it enabled IWDG, but I commented out the "iwdg_feed()" and the firmware was not restarted.
```
</**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */

/**
 * HAL for stm32duino.com based on Libmaple and compatible (STM32F1)
 */

#ifdef __STM32F1__

#include "../../inc/MarlinConfig.h"

#if ENABLED(USE_WATCHDOG)

#include <libmaple/iwdg.h>
#include "watchdog.h"

void HAL_watchdog_refresh() {
  #if DISABLED(PINS_DEBUGGING) && PIN_EXISTS(LED)
    TOGGLE(LED_PIN);  // heartbeat indicator
  #endif
  // iwdg_feed();
}

void watchdogSetup() {
  // do whatever. don't remove this function.
}

/**
 * @brief  Initialized the independent hardware watchdog.
 *
 * @return No return
 *
 * @details The watchdog clock is 40Khz. We need a 4 seconds interval, so use a /256 preescaler and 625 reload value (counts down to 0)
 */
void watchdog_init() {
  #if DISABLED(DISABLE_WATCHDOG_INIT)
    iwdg_init(IWDG_PRE_256, STM32F1_WD_RELOAD);
  #endif
}

#endif // USE_WATCHDOG
#endif // __STM32F1__
>
```

sjasonsmith, 2021-01-24:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Some changes went in a while ago to fix Watchdog usage with maple. I don't recall whether they are shipped in 2.0.7.2 or are currently only in `bugfix-2.0.x`.

You can also test the watchdog by enabling `MARLIN_DEV_MODE` then sending `D100` to the firmware.

If you still have problems with the bugfix branch, please be sure to attach your configuration files, as requested in the issue template.

mixware011, 2021-01-25:
Thanks.
I test the "bugfix-2.0.x branch".

mixware011, 2021-01-25:
Thanks.
I test the "bugfix-2.0.x branch".
I commented out the "iwdg_feed()" and the firmware can be restarted.

closed 2021-01-25 by mixware011