MarlinFirmware/Marlin #20847: have error when uncomment  //#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Dmutro85, 2021-01-22:
#### Additional Information
Have error when uncomment  //#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
I did a test on bugfix-2.0.x, the error was repeated.
I can reproduce the problem with a minimum of options enabled.
Just uncomment:
//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

OS - Manjaro
Arduino IDE - 1.8.13
U8glib - 1.19.1
U8glib - 1.18.0
Marlin 2.0.72
Marlin bugfix 2.0.x
Arduino Mega 2650
Ramps 1.4
LCD - RepRap Discount FULL GRAPHIC Smart Controller (https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller)

### Configuration Files
Links for `Configuration.h` and `Configuration_adv.h`:
https://drive.google.com/drive/folders/1nrAklOEjLNJEuB4wNV0whutdcoYeMQK2?usp=sharing
In my config need to be deleted "_MyConf"

### Steps to Reproduce

1. git clone https://github.com/MarlinFirmware/Marlin.git
2. Load "Marlin.ino" in Arduino
3. uncomment  //#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
4. Press "Ctrl+R" to compilling
5. Compiling end with error
	
**Expected behavior:**
I expected "Done compiling"

**Actual behavior:**

/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_outData'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_outClock'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_bitData'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_bitClock'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_bitNotClock'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/avr-ld: libraries/U8glib/clib/u8g_com_arduino_sw_spi.c.o (symbol from plugin): in function `u8g_outData':
(.text+0x0): multiple definition of `u8g_bitNotData'; sketch/src/HAL/AVR/u8g_com_HAL_AVR_sw_spi.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
Using library U8glib at version 1.19.1 in folder: /home/dimon/Arduino/libraries/U8glib 
exit status 1
Error compiling for board Arduino Mega or Mega 2560.

#### Additional Information

https://drive.google.com/drive/folders/1nrAklOEjLNJEuB4wNV0whutdcoYeMQK2?usp=sharing

[image]
[image]
[image]
[image]
[Marlin_github_Uncomment #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.zip](https://github.com/MarlinFirmware/Marlin/files/5855968/Marlin_github_Uncomment.define.REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.zip)
[Marlin_My_Config_Uncomment #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.zip](https://github.com/MarlinFirmware/Marlin/files/5855969/Marlin_My_Config_Uncomment.define.REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER.zip)

ellensp, 2021-01-22:
Use platformio under vscode 
Arduino IDE is to primitive to build large projects like marlin any more.
https://marlinfw.org/docs/basics/install_platformio_vscode.html

closed 2021-01-22 by ellensp

Dmutro85, 2021-01-22:
Thank you friend! I did!