MarlinFirmware/Marlin #20850: [BUG] Unable to compile last commit with HOST_ACTION and G29_RETRY_AND_RECOVER

tomthemod, 2021-01-22:
### Bug Description

Unable to compile 2021-01-22 commit version (no issue with 2021-01-18)
In src/gcode/gcode.cpp, host_action (line 215 / 224) and host_action_cancel (line 231) isn't recognized

G29_RETRY_AND_RECOVER And  HOST_ACTION enabled

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5856891/Configuration.zip)

### Steps to Reproduce

Using Vscode with PlatformIO (updated), and last commit (2021-01-22),
using default config, modify it :
in configuration.h :
- enable BL_TOUCH
- enable AUTO_BED_LEVELING_BILINEAR
- enable Z_SAFE_HOMING
in configuration_adv.h :
- enable HOST_ACTION_COMMAND
- enable G29_RETRY_AND_RECOVER

and compile

**Expected behavior:**
To compile without error

**Actual behavior:**

Error at compilation : 

Compiling .pio\build\LPC1768\src\src\gcode\geometry\M206_M428.cpp.o
Marlin\src\gcode\gcode.cpp: In static member function 'static void GcodeSuite::event_probe_recover()':
Marlin\src\gcode\gcode.cpp:215:7: error: 'host_action' was not declared in this scope
  215 |       host_action(PSTR(ACTION_ON_G29_RECOVER));
      |       ^~~~~~~~~~~
Marlin\src\gcode\gcode.cpp: In static member function 'static void GcodeSuite::event_probe_failure()':
Marlin\src\gcode\gcode.cpp:224:7: error: 'host_action' was not declared in this scope
  224 |       host_action(PSTR(ACTION_ON_G29_FAILURE));
      |       ^~~~~~~~~~~
Marlin\src\gcode\gcode.cpp:231:9: error: 'host_action_cancel' was not declared in this scope
  231 |         host_action_cancel();
      |         ^~~~~~~~~~~~~~~~~~
*** [.pio\build\LPC1768\src\src\gcode\gcode.cpp.o] Error 1

#### Additional Information

GMagician, 2021-01-22:
@tomthemod give a try to PR I posted

tomthemod, 2021-01-22:
Thanks, will try it tomorrow.

PR #20852 (2021-01-22): Fix HOST_ACTION  & G29 compile error
Should fix #20850
closed 2021-01-22 by thinkyhead