MarlinFirmware/Marlin #20854: Erreur de Compilation Module Motion

LE-BLANC-Dom, 2021-01-22:
Bonsoir,
La compilation génère une erreur.
Firmware : Marlin 2.0.7.2
Machine Ender-2 Pro 4.2.7
Board : CREALITY V427
Architectures STM32F1
Environments : STM32F103RET6_creality

Erreur : 

Marlin\src\module\motion.cpp:513:100: error: 'HOMING_FEEDRATE_Z' was not declared in this scope

Resultat

*** [.pio\build\STM32F103RET6_creality\src\src\module\motion.cpp.o] Error 1
================================================================================================= [FAILED] Took 268.10 seconds ================================================================================================= 

Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:04:28.101
============================================================================================= 1 failed, 0 succeeded in 00:04:28.101 =============================================================================================

Par avance Merci de votre réponse

Dominique LE BLANC

ellensp, 2021-01-22:
Please attach your configuration files.

LE-BLANC-Dom, 2021-01-22:
Bonsoir,

 

Je vous communique les fichiers de configuration que j’utilise

 

Se sont ceux que j’ai téléchargés sans avoir effectuer aucune modification de ma part

 

Cordiales Salutations

 

Dominique LE BLANC

ellensp, 2021-01-22:
From where?  We need configuration files to test this, otherwise we are just guessing.

LE-BLANC-Dom, 2021-01-22:
Du site marlin
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5858833/Marlin.zip)

ellensp, 2021-01-22:
this is the wrong version. it has CONFIGURATION_H_VERSION 020008, this is for bugfix.  You need CONFIGURATION_H_VERSION 020007

Use this https://github.com/MarlinFirmware/Configurations/archive/release-2.0.7.2.zip

closed 2021-01-22 by ellensp

LE-BLANC-Dom, 2021-01-22:
Bonne soirée.
Ca marche merci de cette assistance rapide et efficace.

Cordiales Salutations
Dominique LE BLANC