MarlinFirmware/Marlin #20857: [FR] Add this Creality Ender 2 with Creality Silent Board and stock mini LCD screen configuration

cjj25, 2021-01-23:
### Description

I want to share a working configuration for Marlin that uses the stock mini LCD screen. This took hours of research of the U8GLIB library and I'm hoping someone could add it to the source (tested).

There is already an option for "ENDER2_STOCKDISPLAY" but it only appears to work on the stock Ender 2 board. 

This configuration below works for the stock Ender 2 screen with the Creality Silent Board v1.1.5.

### Feature Workflow

1. Add the code below into /Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
2. Test it's not breaking other configurations

#### Additional Information
```
#if ENABLED(ENDER2_STOCKDISPLAY)
  #undef DOGLCD_CS
  #undef DOGLCD_SCK
  #undef DOGLCD_MOSI
  #undef DOGLCD_CS
  #undef U8G_CLASS
  #undef U8G_PARAM

  #define DOGLCD_SCK BEEPER_PIN // 27
  #define DOGLCD_MOSI LCD_PINS_ENABLE // 17
  #define DOGLCD_CS LCD_PINS_RS // 28
  #define U8G_CLASS U8GLIB_MINI12864_2X
  #define U8G_PARAM DOGLCD_SCK, DOGLCD_MOSI, DOGLCD_CS, DOGLCD_A0
  #define LCD_CONTRAST       1

  #undef BOARD_ST7920_DELAY_1
  #undef BOARD_ST7920_DELAY_2
  #undef BOARD_ST7920_DELAY_3
  #define BOARD_ST7920_DELAY_1 DELAY_NS(200)
  #define BOARD_ST7920_DELAY_2 DELAY_NS(200)
  #define BOARD_ST7920_DELAY_3 DELAY_NS(200)

#endif
```

ellensp, 2023-04-29:
Just added real ender 2 board and found the exp1 has hardware spi pins, this is why  ENDER2_STOCKDISPLAY works on ender 2 controller and not the melzi_creality board. it is still trying to use hardware sck and mosi pins.

ellensp, 2023-04-30:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25759

closed 2023-04-30 by ellensp