MarlinFirmware/Marlin #20862: [BUG] Marlin/src/MarlinCore.cpp:1292:1: error: expected '}' at end of input

CRCinAU, 2021-01-23:
Attempting to build for the CR10-V2 with the latest bugfix-2.0.x.

Configs:
https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/CR-10%20V2

```
In file included from Marlin/src/inc/../HAL/./AVR/HAL.h:24:0,
                 from Marlin/src/inc/../HAL/HAL.h:30,
                 from Marlin/src/inc/MarlinConfig.h:31,
                 from Marlin/src/MarlinCore.h:24,
                 from Marlin/src/MarlinCore.cpp:31:
Marlin/src/MarlinCore.cpp: In function 'void setup()':
Marlin/src/inc/../HAL/./AVR/fastio.h:114:31: error: expected ';' before 'do'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                               ^
Marlin/src/MarlinCore.cpp:957:5: note: in expansion of macro 'OUT_WRITE'
     OUT_WRITE(SUICIDE_PIN, !SUICIDE_PIN_INVERTING);
     ^
Marlin/src/MarlinCore.cpp:1292:1: error: expected '}' at end of input
 }
```

ellensp, 2021-01-23:
missing semicolon on end of   https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/MarlinCore.cpp#L956 
`SETUP_LOG("SUICIDE_PIN")`

ellensp, 2021-01-23:
got broken in commit https://github.com/MarlinFirmware/Marlin/commit/c0870d417a68ff4303100f165282d41be9129a5c

ellensp, 2021-01-25:
merged, closing

closed 2021-01-25 by ellensp