MarlinFirmware/Marlin #20865: Unwanted X movement when ABL_BILINEAR_SUBDIVISION is enabled.

ahmetcemturan, 2021-01-23:
### Bug Description

When moving in X axis the Z axis makes a sudden jerk when it is passing the 4th (I measure a grid of 5x5 points) probe point.
It is not affected by Extrapolate beyond grid.  Everything works fine when ABL_BILINEAR_SUBDIVISION is disabled.

This is the same as https://github.com/MarlinFirmware/Marlin/issues/14963#.  I don't know why it was marked as fixed.

### Configuration Files

*//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING

/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
//#define DEBUG_LEVELING_FEATURE

#if ANY(MESH_BED_LEVELING, AUTO_BED_LEVELING_BILINEAR, AUTO_BED_LEVELING_UBL)
  // Gradually reduce leveling correction until a set height is reached,
  // at which point movement will be level to the machine's XY plane.
  // The height can be set with M420 Z<height>
  #define ENABLE_LEVELING_FADE_HEIGHT

  // For Cartesian machines, instead of dividing moves on mesh boundaries,
  // split up moves into short segments like a Delta. This follows the
  // contours of the bed more closely than edge-to-edge straight moves.
  #define SEGMENT_LEVELED_MOVES
  #define LEVELED_SEGMENT_LENGTH 5.0 // (mm) Length of all segments (except the last one)

  /**
   * Enable the G26 Mesh Validation Pattern tool.
   */
  #define G26_MESH_VALIDATION
  #if ENABLED(G26_MESH_VALIDATION)
    #define MESH_TEST_NOZZLE_SIZE    0.4  // (mm) Diameter of primary nozzle.
    #define MESH_TEST_LAYER_HEIGHT   0.2  // (mm) Default layer height for the G26 Mesh Validation Tool.
    #define MESH_TEST_HOTEND_TEMP  240    // (°C) Default nozzle temperature for the G26 Mesh Validation Tool.
    #define MESH_TEST_BED_TEMP      60    // (°C) Default bed temperature for the G26 Mesh Validation Tool.
    #define G26_XY_FEEDRATE         40    // (mm/s) Feedrate for XY Moves for the G26 Mesh Validation Tool.
    #define G26_RETRACT_MULTIPLIER   1.0  // G26 Q (retraction) used by default between mesh test elements.
  #endif

#endif

#if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define GRID_MAX_POINTS_X 5
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  // Probe along the Y axis, advancing X after each column
  //#define PROBE_Y_FIRST

  #if ENABLED(AUTO_BED_LEVELING_BILINEAR)

    // Beyond the probed grid, continue the implied tilt?
    // Default is to maintain the height of the nearest edge.
    #define EXTRAPOLATE_BEYOND_GRID

    //
    // Experimental Subdivision of the grid by Catmull-Rom method.
    // Synthesizes intermediate points to produce a more detailed mesh.
    //
    //#define ABL_BILINEAR_SUBDIVISION  //ACT
    #if ENABLED(ABL_BILINEAR_SUBDIVISION) 
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 3
    #endif

  #endif

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. enable AUTO_BED_LEVELING_BILINEAR
2. enable ABL_BILINEAR_SUBDIVISION
3. execute ABL
4. Move towards and over the 4th probeposition in x

daleckystepan, 2021-01-23:
Try latest bugfix-2.0.x branch. Should be already fixed in #20122.

ahmetcemturan, 2021-01-23:
https://github.com/ahmetcemturan/Marlin/tree/Sapphire-Pro_MKS-Nano-1.2-3.5inch-w-Probe  
it is the latest...

daleckystepan, 2021-01-23:
Unfortunately your version does not seem to be latest bugfix. It doesnt include fix https://github.com/MarlinFirmware/Marlin/pull/20143/files

sjasonsmith, 2021-01-24:
Closing, since this issue has already been fixed in the `bugfix-2.0.x` branch. Please reply if you still have problems after updating to `bugfix-2.0.x`.

closed 2021-01-24 by sjasonsmith