MarlinFirmware/Marlin #20867: [FR] Guide to SD Formatting is required

filimonic, 2021-01-23:
Currently when I format SD card, I often fail using writing to it:
`card.openFileWrite(...) ` goes OK, but `card.write(...)` returns `-1`

Sometimes I get success on `card.write(...)`, but when I switch card back to pc, I see zero-sized files.

I think that FAT16 and FAT32 support is somewhere broken in Marlin or requires some case of specialization.

Can you help me with formatting SD card propertly? I tried [SDFormatter](https://www.sdcard.org/downloads/formatter/),  
HP USB Disk Storage Format Tool and default windows 10 format tool.

Sebazzz, 2021-01-24:
It might also be your board - and not directly a Marlin issue.

At the CR-6 community firmware fork of Marlin we've seen several users where Marlin just formats the SD card. Literally. They stick it in the printer, run a print, and the SD card is completely cleared. All files are gone.

I'm pretty convinced that that is an artifact of Creality bad v4.5.2 board in combination with some SDIO errata of the STM32 processor on that board. It does not seem to happen on the BTT SKR CR6 board.

ellensp, 2021-01-24:
Not related to marlin firmware. If you want this create in https://github.com/MarlinFirmware/MarlinDocumentation

closed 2021-01-24 by ellensp