MarlinFirmware/Marlin #20871: Cant compile 2.0.x with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

rodribike, 2021-01-24:
### Bug Description

While compiling 2.0.x i get and error:  Marlin\src\gcode\feature\advance\../../../module/stepper/../../feature/../lcd/../module/../libs/duration_t.h:24:10: fatal error: ../HAL/shared/Marduino.h: No such file or directory

i have probing error with version 2.0.7.2 therefore i try this bugfix. i opened both versions and i changed the new bugfix file with same values has 2.0.7.2 (yes some stuff are different and i did not change those)

if i comment REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER VScode can compile...

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step] try to compile firmware 2.0.x with REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER uncommented
2. [Second Step] error shows up
3. [and so on...]  comment REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
4. compile --> success
5. uncomment again
6. error

**Expected behavior:**

success compile with  REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 

**Actual behavior:**

error while compiling
[image]
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5862343/configs.zip)

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-24:
Please don't upload .rar files with zip extensions.

ellensp, 2021-01-24:
The config files is clearly for bugfix branch of marlin not 2.0.7.2

ellensp, 2021-01-24:
builds fine on marlin bugfix.

closed 2021-01-24 by ellensp

sjasonsmith, 2021-01-24:
I also had built it fine. I had delayed commenting while I investigated (and fixed) the LiquidCrystal-related CI failures.

rodribike, 2021-01-24:
> I also had built it fine. I had delayed commenting while I investigated (and fixed) the LiquidCrystal-related CI failures.

What did you fix? Maybe I need to fix too
BTW I changed directory and now works. I'm having issues with long directories...