MarlinFirmware/Marlin #20877: [FR] (feature request title)

Doyle4, 2021-01-25:
Hello,

Will the Raspberry Pi Pico be supported in the future? 

Thanks :)

thawkins, 2021-01-28:
This beast has 16 pwm and pulse train generators built in,  it should be capible of very high step rates.

ellensp, 2021-01-31:
A nice fairly unbiased compassion to other commonly available chips 
https://www.youtube.com/watch?v=cVHCllbN3bQ

panchalnikunj, 2021-02-13:
https://www.youtube.com/channel/UCA4j-J-TFW_9Dt4iTwIjtYQ

Once visit this channel for raspberry pi pico projects.

KillerBug, 2021-04-12:
I built one.  It's a Delta with a 3-in-1-out effector...so 6 stepper drivers.  Running the thermistors into the stock ADC to see if it's good enough.  If not, there is an i2c header that could be used for a different ADC (and maybe a pendant).

It's on a custom board...it has a few issues but I don't want to spin a new one before I know if the stock ADC will work (if not, I'll include something onboard in the next revision).

At the moment my problem is getting marlin to compile for it.  The people over at the marlin forum say I need a new HAL, but I have no idea how to even start with that.

Happy to alpha/beta test.

Here's the post from the Marlin forum with my config files: https://reprap.org/forum/read.php?415,882792

https://ibb.co/tKhKm8V

Hedda, 2021-06-09:
PlatformIO at least now look have platformio platform-raspberrypi suppport for RP2040 on Raspberry Pi Pico and Arduino Nano RP2040 Connect boards

https://github.com/platformio/platform-raspberrypi

https://github.com/platformio/platform-raspberrypi/blob/develop/boards/pico.json

https://github.com/platformio/platform-raspberrypi/blob/develop/boards/nanorp2040connect.json

https://docs.platformio.org/en/latest/platforms/raspberrypi.html

https://docs.platformio.org/en/latest/boards/raspberrypi/nanorp2040connect.html

https://docs.platformio.org/en/latest/boards/raspberrypi/nanorp2040connect.html

https://docs.platformio.org/en/latest/boards/#raspberry-pi-rp2040

https://docs.platformio.org/en/latest/boards/raspberrypi/pico.html

https://docs.platformio.org/en/latest/boards/raspberrypi/nanorp2040connect.html

CaptainNico54, 2021-08-10:
I am looking at developing a new board using the RP2040 (Pi Pico) and I am looking for Marlin support. 

Now that the MCU is fully supported by Arduino and Platform.io, what would we need to do to get the Marlin team to add support for this?

thinkyhead, 2021-08-14:
Here is a basic starting-point that I "quickly" cobbled together… https://github.com/thinkyhead/Marlin/tree/bf2_RP2040_target_PR

The HAL code is simply a copy of `HAL/STM32` with slight modifications. I picked `HAL/STM32` because Malyan M200 has an ARM Cortex-M0 and uses `HAL/STM32` so it ought to be pretty close. Pretty much all the functions. methods, and macros in `HAL/RP2040` will need to be modified with code appropriate to the RP2040.

pkElectronics, 2021-08-29:
Based on the previous work from @thinkyhead and @p3p I started implementing the required macros and functions on my own fork.

If someone want´s to have a testdrive have a look at https://github.com/pkElectronics/Marlin/tree/pr_RP2040
But beware, it´s absolutely work in progress and probably will not yet work as expected.

pkElectronics, 2021-11-01:
Alright, I wanted to give a short update on this.

I finished the implementation of the most basic Marlin subsystems (Analog, Timers, Watchdog) and some more advanced (TMC Uart, SoftSerial). It compiles and is able to move steppers on a rp2040 based control board. If someone wants to have a look please check the branch I provided above. 

I´m not yet sure if the way I implemented the timer subsystem is the best approach to this topic but by now it´s the best one I came up with. The timing of the stepper pulses definetly needs some verification if this is as accurate as necessary.

terjeio, 2021-12-05:
> The timing of the stepper pulses definetly needs some verification if this is as accurate as necessary.

You may use [PIO code](https://github.com/grblHAL/RP2040/blob/master/driverPIO.pio) for pretty accurate timing.

thinkyhead, 2022-01-01:
> … If someone [wants] to have a testdrive …

I got pretty far in compiling the code, rebased with the current bugfix-2.0.x but it stalls because I need to install LVGL someplace. I assume it'll build for a Linux-like target.

> You may use PIO code for pretty accurate timing.

We will seldom need to send pulses in the way that this facilitates. The pulses out of Marlin are tied to the Bresenham algorithm, so they are interleaved in an irregular manner, and spend a majority of the time in acceleration / deceleration.

AlexFroio, 2022-01-22:
Adding onto this thread so I don't make a new one, but a pico based board has been released by BIQU:
[BTT SKR Pico v1.0](https://www.biqu.equipment/products/btt-skr-pico-v1-0) 

It's aimed at Klipper but can still benefit from a Marlin port.

This may make it easier to test on actual hardware without making custom boards.

thinkyhead, 2022-01-24:
> a pico based board has been released by BIQU

I have it here and I am working on getting Marlin installed onto it.

thinkyhead, 2022-01-26:
> It's aimed at Klipper but can still benefit from a Marlin port.

I see one immediate limitation is there's no supported way to attach a common LCD controller, so you need to use OctoPrint or some other host to get an interface. The provided demo HAL for RP2040 has a dubious dependency on LVGL, so I'm working on removing that and getting it to the point of having a working console. At that point I'll post a PR for further testing.

pkElectronics, 2022-01-27:
> The provided demo HAL for RP2040 has a dubious dependency on LVGL

Which HAL are you currently working with? I´m pretty sure I have neither installed nor used LVGL anywhere

thawkins, 2022-01-27:
You could create a separate display system around a second pico or other
soc that attaches via serial,  that approach is used on grbl systems to add
a control dongle with sdcard reader. The hardware for that already exists,
just needs alternative firmware.

https://www.ebay.ph/itm/GRBL-Offline-Hand-Controller-for-DIY-CNC-1610-2418-3018-Laser-Engraving-Machine-/273349597067?mkcid=16&mkevt=1&_trksid=p2349624.m2548.l6249&mkrid=711-127632-2357-0

The interface is literally just RX, TX, GND and VCC.

ellensp, 2022-01-27:
@thawkins most touch displays on 3d printers are serial devices...    we are not short on serial display options.

Hedda, 2022-01-28:
> most touch displays on 3d printers are serial devices... we are not short on serial display options.

@bigtreetech Does Biqu have any specific ideas, suggestions or recommendations for screens/displays + controller system or touchscreens with serial interface for BTT SKR Pico if would use it as stand-alone with Marlin Firmware without a Raspberry Pi?

AlexFroio, 2022-01-28:
You could technically use the laser port for the touch screen yeah. RST just doesn't have anywhere to go to on the board (but that's a hardware problem, not Marlin's) and it can't run an LCD interface.
IO0 and IO1 are both UART capable. The SKR Pico schematic even labels them as TXD0 and RXD0, and they only hook up to this port, and are unused otherwise.
[image]

DottiestDuke47, 2022-04-14:
Has anyone begun any work on a HAL for this platform, I wanted to experiment by adapting/writing my own, but it would be easier if somesones already laid some groundwork. If not, what exactly does a HAL need to incorperate, I've had a look at avrious HAL.cpp files and https://marlinfw.org/docs/development/hal.html, I just want to know if theres a sort of checklist to what needs to be done.

thisiskeithb, 2022-04-14:
> Has anyone begun any work on a HAL for this platform

Yes: https://github.com/MarlinFirmware/Marlin/issues/20877#issuecomment-898782712

thisiskeithb, 2022-04-15:
Closing since there’s now a PR: https://github.com/MarlinFirmware/Marlin/pull/24042

closed 2022-04-15 by thisiskeithb