MarlinFirmware/Marlin #20881: [BUG] Fysetc mini 12864 v2.1 - background LED turns off after bed PID tune

bipsendk, 2021-01-25:
### Bug Description

Fysetc Mini 12864 v2.1 neopixel display turns off background light after bed PID autotune.

### Configuration Files

Attached
[Conf-12864-BedTune.zip](https://github.com/MarlinFirmware/Marlin/files/5869218/Conf-12864-BedTune.zip)

### Steps to Reproduce

1. Configure marlin with attached files
2. Run autotune on bed : M303 E-1 S75 C5
3. When PID autotune ends, the background LED of the display is turned off.

**Expected behavior:**

Display background LED backlight would still be on.

**Actual behavior:**

When PID autotune ends, the background LED of the display is turned off.

#### Additional Information

bipsendk, 2021-01-25:
Observed on 2.0-bugfix pulled 20210124

bipsendk, 2021-01-26:
Just tried extruder hotend PID tune - the backlight LED is switched off here as well once the PID autotune has completed.

3Duke3, 2021-01-26:
I can confirm this, happend on hotend PID tune for me.

ellensp, 2021-01-26:
@3Duke3 can we get your configs also?

3Duke3, 2021-01-26:
Bare with me, new to Github. Not new to Marlin, but certainly no expert.

I am having seperate troubles with behaviour of tool changes on Dual_X but noticed the screen go black on PIDautotune the other day.

While screen is black you can faintly see the letters and head to display - turn on LED

hows best to attach files, zipped like this?

[CONFIG.zip](https://github.com/MarlinFirmware/Marlin/files/5876775/CONFIG.zip)

closed 2021-03-19 by github-actions[bot]