MarlinFirmware/Marlin #20883: [FR] Extended Options for Dual-X Carriage Printers

mikejr83, 2021-01-25:
### Description

I'm not sure if this a bug, configuration, or feature request (or slicer settings). I've been looking at the configuration files and don't see how to do what I'm going.

With IDEX printers, those printers with dual-x carriages, that are set up to change from one tool to another `T0` or `T1` in g-code, would be advantageous to have the newly active tool/carriage ready to go before moving over the build area. In otherwords, the print is currently working with E1 (E0 or T0) and the g-code now calls for a tool change to E2 (E1 or T1). The nozzle should heat and if necessary prime itself before entering the build area without the need for specific g-code.

### Feature Workflow

1. E1 (E0 or T0) is actively in use
2. The g-code encounters a tool change to E2 (E1 or T1)
3. Based on an option that was flagged in the firmware or via a setting/configuration in EEPROM the nozzle heats to the desired temperature, and purges material.
4. The tool then moves to the place where the print will resume
5. Printing immediately resumes. 

#### Additional Information

At the moment the tool change happens and the carriage is moved to the point where the print will resume but it waits for some time. During this time the nozzle may ooze material.

**How is this different than TOOLCHANGE_FILAMENT_SWAP?**

When this is defined in the firmware `TOOLCHANGE_FS_EXTRA_PRIME` can be set to prime the nozzle. However, it still appears that most of the time the carriage will still return to the resume point and wait. I have been unable to figure out the logic as sometimes it appears to immediately resume while other times it still waits. 

The whole goal for this functionality would be to avoid using a prime tower and have the standby extruder sit in a ready and resting state (lower nozzle temperature with filament retracted) and when it becomes active to raise the temperature and prime before moving over the build area and when it does it immediately has the print resume.

mikejr83, 2021-01-25:
It didn't occur to me to post what the slicer is doing at the tool change so here it is:

```
;LAYER:12
G1 F1200 E78.14801
G1 F600 Z3.7
G92 E0
T1
G92 E0
M105
M109 S205
M104 T0 S0
;MESH:cut_1.stl
G0 F7200 X161.9 Y140.1 Z3.7
M104 S215
;TYPE:WALL-INNER
G1 F600 Z2.7
G1 F1200 E16
G1 F3600 X169.4 Y140.1 E16.24945
```
After re-reading it several times it appears that after the tool change the slicer is instructing T0 to cool down (it is no longer used in the print). Then it moves the current tool into position and waits. This may not be an issue with Marlin after all. Can someone confirm that I'm reading this correctly?

almhfysl284, 2021-01-25:
4147241005592098

mikejr83, 2021-01-25:
> 
> 
> 4147241005592098

What?

bryanpedini, 2021-01-26:
> What?

[image]

mikejr83, 2021-01-26:
Wonderful. Anyhow, I'm going to close this issue as I tested a model sliced in PrusaSlicer and it didn't perform in the same manner. Sorry for the distraction devs!

closed 2021-01-26 by mikejr83