MarlinFirmware/Marlin #20888: [FR] Gcode looping (kinda like old "goto")

bryanpedini, 2021-01-26:
### Description

Would it be possible to add a `mark <name>` and a `goto [<line>/<mark>]` functions, to create conditional loops or something like that for continous production, or even let the slicers leverage the functions for a cube with linear infill or similar simple geometries?

### Feature Workflow

1. initial calibration
2. set mark
3. print object
4. lower bed temperature
5. use the printhead to gently remove the object and let it drop in a box or something similar
6. goto mark (repeat)

#### Additional Information

Just a feature question - nothing immediate for sure.

qwewer0, 2021-01-26:
Like [M808](https://marlinfw.org/docs/gcode/M808.html)?

bryanpedini, 2021-01-26:
> 
> 
> Like [M808](https://marlinfw.org/docs/gcode/M808.html)?

Thanks for the quick response and suggestion!
Yes, kinda, actually. I was thinking more of "real markers", like named markers you can use to jump between like old Assembly code or Windows batch files, but that would (mostly) do the trick 😅

closed 2021-04-15 by thinkyhead