MarlinFirmware/Marlin #20890: [DISCUSSION BEFORE PR/Need expertise] (Combined infills / Linear advance codependency)

studiodyne, 2021-01-26:
When using combined infills , for ex : Outlines in 0.1mm in 4 passes , and Infills in 0.4mm in one pass (for gain of time )

An issue appears , the machine doesn't have the ' linear advance setting for the 0.4mm layer height ' and each lines in 0.4 are shorter , don't overlap , and infills are SHIT ! . If i use only one linear advance setting , either i have big corners or i have unglued infills... Linear Advance not ' really linear in vertical mode 😄', because depending of the nozzle size, layer height , direct drive or bowden , solid filament or flexible , two many parameters in the equation 😢

My idea :
Find the ' current layer height ' and change the linear advance .
How to find the current layer height ? By calculating the flow ' difference ' , if the flow is 2x or 4x higher we have the current layer height , and we can apply the change by using an ' array of each value by layer height'. We only use ' multiple 2x/4x/6x ' for better detection of combined infills and to avoid some changes in ' wall single infill '

Is it possible or something will come to interfere or prevent it from working?
Thks

XDA-Bam, 2021-01-30:
OK, so, this is tricky. The relevant info we have for each block/segment is steps per axis (X,Y,Z,E) and the calibrated steps/mm per axis. Thereby, the planner knows the ´extruded length/movement length´. But that's also really all we know - there is no "baseline" or reference value for this ratio. Let's say we define one and then also define some sort of function which change `lin_advance_k` for each block based on this ratio.

We then have the problem that `k` may change drastically at the end of a block in your infill - it could be 0.8 for the thick infill but only 0.5 for the walls or so.  That means, your `advance_steps` are incorrect at the end of the last printed infill block and you will move into the wall with too much nozzle pressure. That creates blobs on the wall, whereever the print head came from an infill line. I'm pretty sure that there's a dozen more such pitfalls. And you can't fix them with the way the Marlin planner is working, because you can only 'look back' on planned blocks, but you can't 'look ahead'.

_My take:_ Don't waste your time on this. It won't work with the current planner code. Tune your LA for nice walls and use workarounds in the slicer to get the infill to print adequate - Increase the wall/infill overlap percentage, print the infill slower or increase the infill flow significantly. You get the idea.

Generic remark: I think "[Feature request]" would be a better title descriptor.

studiodyne, 2021-02-05:
What a nice message, you evoke a dynamic mode, depending on the flow ... But that's not the real way I want to approach this. I would prefer a primitive method, like 'finish a block' 'synchronization planner' and 'new LA' for each new block, but with the difference that I trigger as a switch and not as a permanent dynamic calculation. A really primitive trick to have a little more extrusion when you are in a high flow rate filling. The method is easier and will not pose a permanent math problem. I had two solutions, ask simplify3d or cura to place a 'COMMENT' like 'Current layer Height = xxx', and of course we script this in 'M900 Kxxx' and the problem is solved, but nobody understands me , it is however A REAL BUG! and above all it's a real problem, because the method of fine contours and strong fills, can accelerate the impression x 3. It seems that nobody understands me, and that nobody cares

I have studied what you said, and indeed, very few lines are missed or fail when filling, and I can compensate for this by a slight upper extrusion

Don't wast my time, yes you're right !!!!!!!!!!!!!!!!!

But .... If however CURA or SIMPLIFY3D, adds a tab or a comment to inform the machine about the layer height, then it will be enough just to add 'Planner_synchronize' in the code of the 'M900' and no more problem

thanks a lot

closed 2021-02-12 by studiodyne

thinkyhead, 2021-04-15:
> it will be enough just to add 'Planner_synchronize' in the code of the 'M900'

```cpp
  if (newK != oldK) {
    planner.synchronize();
    kref = newK;
  }
```