MarlinFirmware/Marlin #20891: [FR] Advanced PROBE_OFFSET_WIZARD

KUGA2, 2021-01-26:
### Description

Have an advanced probe offset wizard that also gets x and y

### Feature Workflow

1. Tape a  paper with drawn crosshair to your bed.
2. Manually Move probe over crosshaircross (Confirm in LCD)
2.1.  or repeat if probe missed, or adjust crosshair.
3. Probe moves down till triggered
4. Manually Move nozzle directly on crosshair. (Confirm in LCD)
5. Loosen crosshair paper. Raise nozzle till you can move crosshair paper (Confirm in LCD)
Software can now calculate X, Y and Z from movement.
6. Ask user to confirm storing this settings in eeprom. (Users might forget that in current wizard)

#### Additional Information

Measuring this might be more exact. In step 2 it is quite hard to get right. However x and y is not critical for ABL.
However, Y should be more exact then before, since it is on the same X/Y position (in opposite to normal probe wizard).

fiveangle, 2021-01-26:
Unless one incorporates something like a metal slug that sits above the metal bed that can trigger far earlier than the bed so that it could be measured where the sensor triggers in the XY plane, this "wizard" would be barely better than eyeballing the measurement directly with calipers or even a ruler.  The eye is actually quite good at gauging perspective and parallel views normally.

Prusa designed around this limitation of not knowing exactly where the sensor triggers by embedding special places on the bare PCB bed to locate the bed and combined that with their design knowledge of their extruder + sensor mounting.  But one could design the same in reverse by, for instance, designing for say a known quantity trigger point.  Let's take the US Dime for example...

- Home printer
- 3D print a cylinder with adequate brim to the bed at a known fixed XY offset location that can be used to capture a stack of US Dimes
- Move the gantry higher than the Z location of the stack of dimes you plan to use that is taller than the sensor trigger range so that the sensor will trigger on the stack but not be affected by the surrounding bed
- Fill the printed cylinder on the bed with US Dimes
- Successively move the gantry back and forth in the X direction while successively iterating slowly in the Y direction, building a sensor map where the sensor triggers the trailing and leading edges of the stack of dimes
- Determine the mean average of the center between the detected edges on X axis and set sensor X location to that location minus the fixed X offset of the printed cylinder tower.
- Do the same for the Y axis
- This is the exact sensor offset from the nozzle so save it to EEPROM
- Drink 🍺 

But a purely visual "wizard" seems no better than direct visual measurement because like I said, the eyeball is quite good at doing this already, to within the sensor +/- range tolerances.  Getting any more accurate in that measurement doesn't gain anything other than warm fuzzy feelings: the sensor is still not guaranteed to trigger exactly under that offset, even if it is visually exact to the dimensions of the sensor center.

The empirically measured wizard method I suggest above would be able to be accomplished purely in the slicer + post-processing, although it could certainly be optimized to be easier within the firmware... perhaps as part of the DEBUG switch so it isn't enabled by default (since it would only need to be characterized once for the life of the printer or mount).

qwewer0, 2021-01-26:
One thing came to my mind is that the Z offset can change from nozzle to nozzle, but the XY offset won't change unless something big changes (e.g. fanduct) around the hotend, and that happens a lot less often, or just once in a 3d printer's life.

gutbrod, 2021-01-27:
@KUGA2's comment is spot on, as I have a cartesian (Ender 3 v2) and this is a tinker item for me, I spend weeks tuning this, trying different fan ducts and probe holders. I used the process similar to what he described:
Calculate the Probe X/Y/Z Offset from the nozzle
   a. Homed the printer
   b. Taped a paper with a dot to my bed
   c. Deployed the BL Touch probe tip
   d. Moved the probe tip to the dot
   e. Moved Z "down" (using 1 mm, then .1 mm, then .025 mm increments) until the probe turned red and triggered
   f. Manually recorded the x/y/z from the screen
   g. repeated the Z motion 5 times and averaged the Z numbers
   h. Moved the nozzle to point
   i. Inserted a piece of paper between the nozzle 
   i. Moved the Z "down" until I could feel a piece rubbing between the dot and the nozzle
   j, Repeated the process 5 times to get the average, then subtracted the .1 mm for the paper

@fiveangle Not all of us are gifted with your apparently amazing vision, I have partial blindness in one eye, and need reading glasses (so depth perception can be iffy, especially close up). It also seems like your comment is targeted to Induction probes, but why use a dime to figure out the trigger point of the Inductive sensor and not the bed? If you're using glass as a bed material that would make the dime results useless. I don't see how anything in your comment solves the X/Y issues. You don't want your probe to be off the 0/0 side of the bed, or for the printer to slam into the opposite X/Y hard stop trying to get the probe to check the level of of 220/220 in X/Y. The process would also allow for calculating the safe zone for probing (which Marlin UBL just ignores unless I missed a bugfix notice).

@qwewer0 Many Inductive sensors have a bolt and screw motion to move them closer or farther from the bed, the constant racking motion of the hot end head/fan duct can cause it to move in the bolt and thus result in losing accuracy over time, unless everyone starts using Loctite.

Having this method in firmware would be a benefit for print farms too, allowing them to regularly calibrate their printers maintaining the quality (I have had quite of few items from "printer farms" that were absolutely useless because of the quality of the first few layers).

qwewer0, 2021-01-27:
> @qwewer0 Many Inductive sensors have a bolt and screw motion to move them closer or farther from the bed, the constant racking motion of the hot end head/fan duct can cause it to move in the bolt and thus result in losing accuracy over time, unless everyone starts using Loctite.

For the Z offset calibration we have the present `PROBE_OFFSET_WIZARD`.

tesla128, 2021-02-01:
Seconded purely for the imperfect vision aspect and never assuming everyone can see perfectly to make adjustments.  I can't thread a needle or clean a nozzle to save my life without a microscope

qwewer0, 2021-02-01:
> Seconded purely for the imperfect vision aspect and never assuming everyone can see perfectly to make adjustments. I can't thread a needle or clean a nozzle to save my life without a microscope

This theoretical xy offset wizard would also just rely on the sight, just make procedure semi automatic.

fiveangle, 2021-02-13:
@gutbrod 's unecessary sarcasim notwithstanding, it seems the proposal is to create a visual aid that relies on the exact visual acuity that the proponents of the wizard claim they don't possess and need the aid for.

Regarding my suggestion, yes it was for induction probes.  I would never have assumed you meant other pinpoint style probes such as BLTouch and the like, since they are far easier to visually see the trigger point (it's a point after all).

Regardless, I'm sure that Scott would consider the inclusion of such as wizard if it were coded to be turned off by default. Anything that aids accessibility without impacting other areas of the code should always be welcome.

KUGA2, 2021-04-09:
My original request is mainly about the following issue:

When using the existing PROBE_OFFSET_WIZARD, The Offset is measured in a different Place (X/Y) then the nozzle is. If you bed is wrapped, this might impact on print adhesion.
With my approach, this is measured in the same spot.
(Additionally, the X/Y Offset can be determined better then you could when you have a DIY solution for the sensor holder and have to measure it with a ruler)

qwewer0, 2021-04-09:
> When using the existing PROBE_OFFSET_WIZARD, The Offset is measured in a different Place (X/Y) then the nozzle is. If you bed is wrapped, this might impact on print adhesion.

That shouldn't be the case on the latest bugfix-2.0.x

@KUGA2 Could you write step by step details to what is happening on your end? And could you attach your configuration files?

thisiskeithb, 2021-04-09:
Please download [`bugfix-2.0.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to test with the latest code and let us know if you're still having this issue.

KUGA2, 2021-04-14:
I do not think this is a bug. I am on fairly new bugfix-2.0.x. I think you just misunderstand me. I will record a video for you at weekend.