MarlinFirmware/Marlin #20892: [BUG] MKS Robin nano v.2, CUSTOM_USER_MENUS (in configuration_adv) not working

Malderin, 2021-01-26:
Board: MKS robin nano v.2
Last marlin 2.x bugfix

After activating CUSTOM_USER_MENUS in configuration_adv in the "Tools" menu, the item "more" should appear.
In fact, nothing appears. Three weeks ago, everything worked.
[conf.zip](https://github.com/MarlinFirmware/Marlin/files/5875946/conf.zip)

Malderin, 2021-01-27:
[image]
[image]

rhapsodyv, 2021-01-30:
Seems the "More" menu never existed in official marlin... I think your issue is with the stock firmware.

@makerbase-mks @MKS-Sean @mks-viva can you confirm this? Maybe it's missing PR.

Malderin, 2021-01-30:
@rhapsodyv .
this is a fork of the official marlin 2.x bugfix, where I added pr with mks wifi, where the menu "more" works with the same configs:
https://github.com/Malderin/Marlin_2.0.x_official/tree/Marlin-2.0_bugfix_MKS_WIFI.
I will repeat myself. The more menu worked about 3 weeks ago.

rhapsodyv, 2021-01-30:
You can take a look at the history of the file draw_tool.cpp ... it never had the button "more" on marlin: https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x/Marlin/src/lcd/extui/lib/mks_ui/draw_tool.cpp

rhapsodyv, 2021-01-30:
> I will repeat myself. The more menu worked about 3 weeks ago.

You can repeat yourself as much as you want. It will not help anyone to be right or wrong here.

Malderin, 2021-01-30:
> You can take a look at the history of the file draw_tool.cpp ... it never had the button "more" on marlin: https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.0.x/Marlin/src/lcd/extui/lib/mks_ui/draw_tool.cpp

You may be right, sorry

rhapsodyv, 2021-01-30:
Probably we just need ask MKS to send it to Marlin Official, as it's clearly already done in their repo (or somewhere).

Malderin, 2021-01-30:
I corrected myself. Someone removed the #define USER_CMD_1_ENABLE parameters from the config_adv.n (and 2, 3 ... 7). Therefore, the "more" menu was empty.

> Probably we just need ask MKS to send it to Marlin Official, as it's clearly already done in their repo (or somewhere).

In their repository (MKS) this was done a long time ago. The problem is that their code is very different from the code in the official marlin.

MKS-Sean, 2021-01-31:
@rhapsodyv @Malderin Hello, I will submit a PR to marlin soon to fix some code problems of lvgl

rhapsodyv, 2021-02-02:
Merged

closed 2021-02-02 by rhapsodyv