MarlinFirmware/Marlin #20902: [FR] Add option to M600 to not overwrite the display

ssendev, 2021-01-28:
### Description

When printing parts with multiple colours it's not always easy to tell which colour is next but with custom filament/tool GCode it's possible to have code like `M117 Change Filament to [filament_colour]; M600` unfortunately the message is overwritten with a generic change filament message.

### Feature Workflow

A solution could be if `M600` accepted an option to disable displaying a generic message so that the last `M117` message is still visible after Marlin beeped and is waiting to resume.

ETE-Design, 2021-01-28:
Sound like a great idea, what slicer have already support for that feature?

ssendev, 2021-01-28:
I used PrusaSlicer and just checked Cura. Glancing over the options i couldn't find custom filament G-code but custom Extruder G-code is supported but would then need a postprocessing step to remove the T0 commands. But I do it this way even in Prusa since filament change is only supported on layer change and this way "true" multicolor prints are possible.

Here my custom script for Prusa if anyone is interested:
```G-code
; Tool change G-code
; IGNORE_NEXT_LINE
T[next_extruder]
{if previous_extruder != -1 }M117 Extruder [next_extruder] [filament_colour]
M600
{endif}
```
and then postprocess it with
```bash
#!/bin/bash
sed -i -e '/IGNORE_NEXT_LINE/,+1d' "$1"
```

qwewer0, 2021-01-28:
> M117 Change Filament to [filament_colour]; M600

Maybe you could use `M0` instead of `M117`?
It will display the message and wait for the user, then start the M600 process. Naturally you would need to add a "move away and back" gcode before and after the filament change, so that the nozzle isn't waiting on the printed part and it returns after it.

e.g.
```
G91
G0 Z10 Fxxxx
M0 Change Filament to [filament_colour]
M600
G0 Z-10 Fxxxx
G90
```

ssendev, 2021-01-28:
@qwewer0 I just opened another issue #20910 which could also be solved by having a list of equivalent commands to M600 since the solution you posted is strictly worse than M600 since it doesn't move to the parking position and needs a double confirmation

eoyilmaz, 2023-07-30:
After a playing around with both @ssendev and @qwewer0's recommendations and combining it with a modifier version of [Mike_GKA's answer](https://forum.prusa3d.com/forum/postid/639696/), I found the following Tool change g-code for Prusa slicer is doing exactly what I required:

```
; eliminate first tool change
{if previous_extruder != -1}
; REMOVE_NEXT_LINE
T[next_extruder]
G0 Z{toolchange_z+10} F6000
; color notification
M0 Next color is{if extruder_colour[next_extruder] == "#000000"} BLACK
{elsif extruder_colour[next_extruder] == "#0000FF"} BLUE
{elsif extruder_colour[next_extruder] == "#800000"} BROWN
{elsif extruder_colour[next_extruder] == "#00FFFF"} CYAN
{elsif extruder_colour[next_extruder] == "#999999"} GRAY
{elsif extruder_colour[next_extruder] == "#D4AF37"} GOLD
{elsif extruder_colour[next_extruder] == "#00FF00"} GREEN
{elsif extruder_colour[next_extruder] == "#FF00FF"} MAGENTA
{elsif extruder_colour[next_extruder] == "#FF8000"} ORANGE
{elsif extruder_colour[next_extruder] == "#FF6600"} ORANGE
{elsif extruder_colour[next_extruder] == "#800080"} PURPLE
{elsif extruder_colour[next_extruder] == "#FF0000"} RED
{elsif extruder_colour[next_extruder] == "#C0C0C0"} SILVER
{elsif extruder_colour[next_extruder] == "#F0F0F0"} TRANSPARENT
{elsif extruder_colour[next_extruder] == "#FFFF00"} YELLOW
{elsif extruder_colour[next_extruder] == "#FFFFFF"} WHITE
{endif};
M600
G0 Z{toolchange_z} F6000
{endif}
```

As opposed to @qwewer0's recommendation, my version doesn't use relative positioning which was breaking my Ender 3 extrusion between filament changes.

Also the following code for post-processing scripts removes the ``T#``s which is required:

```
/usr/bin/sed -i -e '/REMOVE_NEXT_LINE/,+1d';
```

And I set the Printer Settings -> General -> Extruders > 1 and then I set the color of each extruder (CMYW for Lithophane prints).

jgbrown54, 2023-07-30:
I do it a little different using Prusa slicer. The slicer is set for multiple extruders even though the printer only has one. I add the following to the Tool Change Custom Gcode.

{if next_extruder!=current_extruder} ; Skip initial Tool Change
; Tool Change Gcode
  {if next_extruder==0}
    M118 Load Extruder With Black Filament
  {elsif next_extruder==1}
    M118 Load Extruder With Blue Filament
  {elsif next_extruder==2}
    M118 Load Extruder With Brown Filament
  {elsif next_extruder==3}
    M118 Load Extruder With Green Filament
  {elsif next_extruder==4}
    M118 Load Extruder With Orange Filament
  {elsif next_extruder==5}
    M118 Load Extruder With Purple Filament
  {elsif next_extruder==6}
    M118 Load Extruder With Red Filament
  {elsif next_extruder==7}
    M118 Load Extruder With Yellow Filament
  {endif}
  M600 B6 E-5 L450 U495 X20 Y280 Z30 ; Change filament
; End Tool Change Gcode
{endif}

The M118 sends a message to the host interface. I just set the proper color name for each extruder. Once it's sliced, the colors are part of the gcode file.