MarlinFirmware/Marlin #20904: [BUG] NEOPIXEL2_SEPARATE causes error with "class LEDLights2" during compile

lightmaster, 2021-01-28:
### Bug Description

Trying to enable `NEOPIXEL2_SEPARATE` causes an error when compiling. (see below for error)

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

<a href="https://drive.google.com/file/d/1kb6WikZRdgHs_nQBoILHMbiUldvBam_L/view?usp=sharing" target="_blank">Marlin Config.zip</a>

### Steps to Reproduce

1. Enable `NEOPIXEL_LED` and `NEOPIXEL2_SEPARATE`
2. Try to compile

**Expected behavior:**

I expect that it would compile successfully

**Actual behavior:**

Compiling errors out with the following error:
```
In file included from Marlin\src\lcd\menu\menu_led.cpp:31:
Marlin\src\lcd\menu\menu_led.cpp: In function 'void menu_led()':
Compiling .pio\build\mks_robin_nano35_stm32\src\src\lcd\tft\images\back_32x32x4.cpp.o
Marlin\src\lcd\menu\menu_led.cpp:130:47: error: 'class LEDLights2' has no member named 'set_default'
  130 |       ACTION_ITEM(MSG_SET_LEDS_DEFAULT, leds2.set_default);
      |                                               ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:261:39: note: in definition of macro '_MENU_INNER_P'
  261 |     MenuItem_##TYPE::action(plabel, ##V);                      \
      |                                       ^
Marlin\src\lcd\menu\menu_item.h:356:57: note: in expansion of macro '_MENU_ITEM_P'
  356 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:368:58: note: in expansion of macro 'MENU_ITEM_P'
  368 | #define ACTION_ITEM_P(PLABEL, ACTION)                    MENU_ITEM_P(function, PLABEL, ACTION)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:369:56: note: in expansion of macro 'ACTION_ITEM_P'
  369 | #define ACTION_ITEM(LABEL, ACTION)                     ACTION_ITEM_P(GET_TEXT(LABEL), ACTION)
      |                                                        ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_led.cpp:130:7: note: in expansion of macro 'ACTION_ITEM'
  130 |       ACTION_ITEM(MSG_SET_LEDS_DEFAULT, leds2.set_default);
      |       ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_led.cpp:130:47: error: 'class LEDLights2' has no member named 'set_default'
  130 |       ACTION_ITEM(MSG_SET_LEDS_DEFAULT, leds2.set_default);
      |                                               ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:266:58: note: in definition of macro '_MENU_INNER_P'
  266 |       (encoderLine == _thisItemNr, _lcdLineNr, plabel, ##V);   \
      |                                                          ^
Marlin\src\lcd\menu\menu_item.h:356:57: note: in expansion of macro '_MENU_ITEM_P'
  356 | #define MENU_ITEM_P(TYPE, PLABEL, V...)                 _MENU_ITEM_P(TYPE, false, PLABEL, ##V)
      |                                                         ^~~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:368:58: note: in expansion of macro 'MENU_ITEM_P'
  368 | #define ACTION_ITEM_P(PLABEL, ACTION)                    MENU_ITEM_P(function, PLABEL, ACTION)
      |                                                          ^~~~~~~~~~~
Marlin\src\lcd\menu\menu_item.h:369:56: note: in expansion of macro 'ACTION_ITEM_P'
  369 | #define ACTION_ITEM(LABEL, ACTION)                     ACTION_ITEM_P(GET_TEXT(LABEL), ACTION)
      |                                                        ^~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_led.cpp:130:7: note: in expansion of macro 'ACTION_ITEM'
  130 |       ACTION_ITEM(MSG_SET_LEDS_DEFAULT, leds2.set_default);
      |       ^~~~~~~~~~~
*** [.pio\build\mks_robin_nano35_stm32\src\src\lcd\menu\menu_led.cpp.o] Error 1
=================================================================================== [FAILED] Took 106.83 seconds ===================================================================================

Environment             Status    Duration
----------------------  --------  ------------
mks_robin_nano35_stm32  FAILED    00:01:46.835
=============================================================================== 1 failed, 0 succeeded in 00:01:46.835 ===============================================================================The terminal process "C:\Users\light\.platformio\penv\Scripts\pio.exe 'run'" terminated with exit code: 1.
```

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

ellensp, 2021-01-28:
As a work around you need to enable NEO2_COLOR_PRESETS in Configuration_adv,h  
But this needs a patch to address this so it errors more gracefully or handles not having   NEO2_COLOR_PRESETS set, so please leave open for now.

lightmaster, 2021-01-28:
@ellensp I have never messed around with Neopixel before, but from the text in the config file, I'm not sure that will achieve the same thing. From my understanding, `NEOPIXEL2_SEPARATE` allows for a second strip of Neopixel to be controlled in parallel with the first strip. `NEO2_COLOR_PRESETS` sounds like its just a second color preset.

ellensp, 2021-01-28:
NEO2_COLOR_PRESETS  is the independent presets for NEOPIXEL2_SEPARATE  
set_default only exists if NEO2_COLOR_PRESETS is set.

lightmaster, 2021-01-28:
Ok, that got it to compile. Thanks.

lightmaster, 2021-01-28:
I'll create another bug report when I wake up, but seems that the first set of Neopixel lights do not change color correctly from the LCD. For instance, click the Blue preset, half the LEDs turn blue, half turn red. I'll definitely look into it more tomorrow and get more details on when they work right vs when they do not all change correctly.

ellensp, 2021-01-29:
I have made a PR to fix this.

closed 2021-01-29 by thinkyhead

lightmaster, 2021-01-29:
Just to confirm, the current `bugfix-2.0.x` 9d0e64a725290233d6002017147c578ffd32c504 does compile for me with `#define NEOPIXEL2_SEPARATE` enabled and `#define NEO2_COLOR_PRESETS` disabled.