MarlinFirmware/Marlin #20905: Compiling error and suggested fix

francko76, 2021-01-28:
Marlin\src\lcd\extui\lib\anycubic_i3mega\anycubic_i3mega_lcd.cpp

row 768:
ExtUI::injectCommands_P(G28_STR);

should be
ExtUI::injectCommands_P(PSTR("G28_STR"));

ellensp, 2021-01-28:
No, that would send "G28_STR" as a gcode, which is invalid.

error is 
```
Marlin/src/lcd/extui/lib/anycubic_i3mega/anycubic_i3mega_lcd.cpp: In member function 'void AnycubicTFTClass::GetCommandFromTFT()':
Marlin/src/lcd/extui/lib/anycubic_i3mega/anycubic_i3mega_lcd.cpp:768:41: error: 'G28_STR' was not declared in this scope
                 ExtUI::injectCommands_P(G28_STR);
```

A more correct solution is to add "extern const char G28_STR[];" to the end of anycubic_i3mega_lcd.h

ellensp, 2021-01-29:
That PR will fix it. 

Next time please follow the bug template when submitting bug reports.

closed 2021-01-29 by thinkyhead