MarlinFirmware/Marlin #20906: [BUG] Latest bugfix commit breaks emergency parser

NovaViper, 2021-01-28:
### Bug Description

I just updated my bugfix branch to the very latest commit (7bbdbcfb6db4c6d35e4aaffca06073dd2bfc6719) and I tried compiling the firmware and immediately got this error

```
Marlin/src/HAL/STM32/MarlinSerial.cpp: In member function 'void MarlinSerial::_rx_complete_irq(serial_t*)':
Marlin/src/HAL/STM32/MarlinSerial.cpp:84:31: error: 'emergency_state' was not declared in this scope; did you mean 'emergency_parser'?
   84 |       emergency_parser.update(emergency_state, c);
      |                               ^~~~~~~~~~~~~~~
      |                               emergency_parser
Compiling .pio/build/BIGTREE_SKR_PRO/src/src/core/utility.cpp.o
*** [.pio/build/BIGTREE_SKR_PRO/src/src/HAL/STM32/MarlinSerial.cpp.o] Error 1
```
I do have emergency paraser enabled.. 

### Configuration Files

Here's a link to my branch: https://github.com/NovaViper/Marlin/tree/xavier-stable2.0.x/Marlin
Aswell as the run: https://github.com/NovaViper/Marlin/runs/1785661931

### Steps to Reproduce

1. Get the 7bbdbcf commit
2. Enable Emergency parser 
3. Compile for an SKR_Pro board
4. The error occurs

**Expected behavior:**

It should compile like normal and throw out the error

**Actual behavior:**

It throws out that error mentioned above

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

040medien, 2021-01-28:
Can confirm the error - also on SKR Pro

Minims, 2021-01-29:
Same problem on with SKR Pro with latest commit

bipsendk, 2021-01-31:
Same issue when trying to compile for FYSETC S6 board

X-Ryl669, 2021-01-31:
Please test with latest bugfix, it should be resolved. If so, please close the issue. Thanks!

NovaViper, 2021-02-02:
Still failing with latest commit: https://github.com/NovaViper/Marlin/runs/1811954270?check_suite_focus=true

It doesn't seem like the fix was applied to Marlin at all: https://github.com/NovaViper/Marlin/commits/xavier-bug2.0.x

X-Ryl669, 2021-02-02:
The errors in your build are not related to this issue, you've configuration issues in your build:
```
TRAMMING_POINT_XY point 0 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
TRAMMING_POINT_XY point 3 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
```

040medien, 2021-02-02:
I can confirm that the fix works.

NovaViper, 2021-02-02:
Ah I fixed the tramming error! And I can confirm the merge fixed the issue. Closing this now

closed 2021-02-02 by NovaViper