MarlinFirmware/Marlin #20907: [BUG] Marlin, ST-Link V2, and Creality's STM32 32bit motherboards

jemzipx, 2021-01-28:
I apologize if this is not the right place to ask this. I have asked this question on Marlin's discord and FB pages among other places and no one knew about the solution. I want to contribute code to Marlin. I have bought a clone ST-Link V2 adapter (from adafruit) to connect to my Ender 3 V2 with V4.2.2 board and 32-bit STM32F103 chipset. I tried to test the connection with ST Link Utility app but it gives me 'cannot connect to the target' error. Since I read online that these st-link clones have wrong labels printed on their shell, I decided to take off the shell and have a look at markings on the PCB. They match. I connected SWCLK, SWDIO, GND to corresponding pins on the 3d printer's board, turned on the 3d printer, and connected the ST-Link adapter. I launch ST-Link utility app but I get this 'cannot connect to the target' all the time. I have played around a lot with lots of settings and hardware variations with and no matter what I do I get the same error. I did this from several operating systems and computer and all of them give me the same error. Since I bought two of these st-link dongles, I tried the other one, same results. I have upgraded their firmware and still no luck. I have played with various 'modes' and settings, no luck. My question is, has anyone here successfully connected ST-Link  to Ender 3 V2? In a linux distro, I get the following:

> sudo st-info --probe
>   version:    V2J37S7
>   serial:     294e040011145157544d4e00
>   hla-serial: "\x29\x4e\x04\x00\x11\x14\x51\x57\x54\x4d\x4e\x00"
>   flash:      0 (pagesize: 0)
>   sram:       0
>   chipid:     0x0748

flash and sram is 0 which is kind of weird I guess. In Windows, I just get the 'cannot connect to the target' in st-link utility all the time:

I am sorry for asking here. I know this is not a bug nor a feature but I could not find a solution elsewhere. Am I missing something? The ST-Link version is 'Mini ST-Link V2 D' (written on the PCB). It uses CKS chipset. 
[image]

[image]
[image]

thisiskeithb, 2021-01-28:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2021-01-28 by thisiskeithb

jemzipx, 2021-01-29:
@thisiskeithb As mentioned in my first post, I had asked this question in several other places including the Facebook and Discord page that you recommended in your post and nobody could help. Asking here was my last resort. I apologized in my post and was hoping maybe someone here could help. Well, since you  shut this down so quickly I guess I am not going to find an answer. Kind of disappointing that someone wants to contribute some much to your codebase and you do not even give him a chance.