MarlinFirmware/Marlin #20909: Problem homing delta printer [BUG]

BirkBinnard, 2021-01-28:
### Bug Description

Problem homing delta printer

### Configuration Files
[Binnard- Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5889613/Binnard-.Configs.zip)

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

Specified Sensorless Homing and Probing, Mixing hotend, 2 extruders, TMC2209 stepper drivers

### Steps to Reproduce

1. Home printer with hotend close to Z = max (top of printer) works OK
2. Home printer with hotend close to Z = min (near print bed) causes hotend to crash into X tower 

**Expected behavior:**
Successful Home from any position

**Actual behavior:**
Hotend crashes into X tower when Home initiated from position close to printbed

#### Additional Information
I'm running the latest BugFix version of Marlin on a delta printer. When I Home the printer it homes the X tower first, then Y, then Z. This works fine when the hotend is close to the tops of all 3 towers. 

But when the hotend is close to the print bed Home fails because the X stepper motor moves the X tower slider up until the hotend crashes into the X tower. So I cut power when this happens.

What should happen is all 3 axes should move up until the first one reaches it's top limit, then the 2nd one, and finally the 3rd one. I thought this setting in Configuration_adv.h would do this, but it doesn't:

#define HOME_Z_FIRST

Is there another parameter that allows deltas to home properly from any position?

BirkBinnard, 2021-01-30:
This bug report is the same problem but was closed even though it did not clearly identify a fix:

https://github.com/MarlinFirmware/Marlin/issues/16736

sjasonsmith, 2021-02-08:
This is not the same as #16736. Maybe the symptom is the same, but not the same root cause. That other issue was specific to the pin definitions for the Trigorilla board, which is not used here.

sjasonsmith, 2021-02-08:
> What should happen is all 3 axes should move up until the first one reaches it's top limit, then the 2nd one, and finally the 3rd one.

This is what it normally does, although I have not actually tried it since December.

> #define HOME_Z_FIRST

This is probably not relevant to delta, since you _only_ home Z.

sjasonsmith, 2021-02-08:
I see you also filed another bug, which seems to be largely a duplicate of this. Could you please close one of them?
https://github.com/MarlinFirmware/Marlin/issues/20986

BirkBinnard, 2021-02-08:
This is a duplicate of #20986 which is currently open, so this one is now closed.

closed 2021-02-08 by BirkBinnard