MarlinFirmware/Marlin #20916: [BUG] (short description)

jeremylms, 2021-01-29:
### Bug Description

Cannot switch off fysetc mini12864 v2.1 neopixel backlight after enabling psu control

### Configuration Files

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5891565/Configuration_adv.zip)

### Steps to Reproduce

1. Connect fysetc mini12864 v2.1
2. Enable NEOPIXEL_LED
3. Hook up a bigtreetech power off relay
4. Define PSU_CONTROL
5. Navigate to the led control menu
6. Switch off lights 

**Expected behavior:**

Backlight will be switched off.

**Actual behavior:**

Backlight doesn't switch off. During print with printer event led enabled or after changing to other colour, switches to default light instead.

#### Additional Information
If psu control is disabled, the backlight will switch off.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/5891565/Configuration_adv.zip)

ellensp, 2021-01-29:
Provided configs are not bugfix compatible have you tried current bugfix?
`#error "HOMING_FEEDRATE_XY and HOMING_FEEDRATE_Z are now set using the HOMING_FEEDRATE_MM_M array instead`

ellensp, 2021-01-29:
[bugfixConfiguration.zip](https://github.com/MarlinFirmware/Marlin/files/5891773/bugfixConfiguration.zip)
updated for current bugfix

jeremylms, 2021-01-29:
Thanks for your help. I have just tried latest version of bugfix but it still does not work.
[Config bugfix latest.zip](https://github.com/MarlinFirmware/Marlin/files/5892607/Config.bugfix.latest.zip)

ellensp, 2021-01-29:
Defiantly something funny going on here. This LCD has 3 neopixels 0 is backlight colour, 1 is left side of knob colour 2 is right side of knob colour.  Even with PSU_CONTROL  LCD control lights off, just turns off rgb 2 , rgb 0 and 1 go red 
Menu show off. With PSU_CONTROL enabled itt seem to change colour a fraction but menu item does not change to off

ellensp, 2021-02-07:
NOTES: all you need to trigger this bug is Marlin and enable
```
#define SERIAL_PORT -1
#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
#define PSU_CONTROL
#define FYSETC_MINI_12864_2_1    // Type A/B. NeoPixel RGB Backlight
```
Tracked it down to Marlin/src/inc/Conditionals_LCD.h
```
#if ENABLED(PSU_CONTROL) && EITHER(FYSETC_MINI_12864_2_0, FYSETC_MINI_12864_2_1)
    #define LED_BACKLIGHT_TIMEOUT 10000
  #end
```
So someone really wanted this set to 10000, but if you comment out #define LED_BACKLIGHT_TIMEOUT 10000, light on/off works again. 

So looking deaper as to why the LED_BACKLIGHT_TIMEOUT 10000?

ellensp, 2021-02-08:
Try this PR.
Please note: When  PSU_CONTROL is enabled and the power supply is off, you cannot manually turn off the backlight. But it is on a 10 second timer to turn off automatically (this is what the LED_BACKLIGHT_TIMEOUT was for, but it also didn't work before)  
PSU_CONTROL is enabled and the power supply is on, The timer is disabled and the on/off now works as expected,

ellensp, 2021-02-08:
this has been merged into bugfix

closed 2021-02-08 by ellensp