MarlinFirmware/Marlin #20917: [BUG] reset button on LCD board does not work for Melzi control board

bill-orange, 2021-01-29:
Bug Description:

On a Monoprice Maker Select V2 with stock board and display, the example config files do not properly setup of the reset button on the LCD board.  The buzzer does not work either, This machine came with a bootloader so it was probably one of the last ones made,

Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5891707/Marlin.zip)

### Steps to Reproduce

1. Press reset button
2. Observe nothing happens
3. Press and hold reset button
4. Observe still nothing

Expected behavior:

Reset as if power was lost or the reset button on the Melzi button was pressed.

Actual behavior:

Nothing happens, no reset

#### Additional Information:

I googled this problem and found an old discussion on Reddit  no one in the tread had a solution.  I suspect it is a pin assignment issues.  If someone can get me started, I can troubleshoot it.

ellensp, 2021-01-29:
It rather difficult to find out the information on theses old machines. but...

these are all the pins that it could be (probably)

```
 *  D12  D11  D10  D17  D16
 *  D31  D30  D29  D28  D27
```
take out the one we know about 
```
    #define LCD_PINS_RS         17
    #define LCD_PINS_ENABLE     16
    #define LCD_PINS_D4         11
    #define BTN_ENC             28
    #define BTN_EN1             29
    #define BTN_EN2             30
```
only the following remain

```
 *  D12   X   D10   X    X
 *  D31   X    X    X   D27
```

So try finding the buzzer with M42, ie try each of these
M42 P31 S1
M42 P10 S1
M42 P27 S1
M42 P12 S255           ;(is pwm capable, so slightly different) 

Set S0 on the same pin to turn off if you find it

bill-orange, 2021-01-29:
@ellensp Thanks, that is the approach I was hoping to take.  I have  two remaining questions however.  There are a lots of different pins files.  How do I tell for sure which one is used in my compile?  I could be changing and testing all day on the wrong one!

The buzzer pin is clearly `#define BEEPER_PIN` in the pins files.  I don't see a define that is clearly the  reset (ESTOP) button.  That's a problem.  Archived resources on my board Identify the Beeper as D27 (apparently incorrect) and the ESTOP as D10.  Where on earth is the reset button pin defined?

ellensp, 2021-01-29:
re pins It all start with #define MOTHERBOARD BOARD_MELZI 
next in pins.h is a list of all boards eg 
#elif MB(MELZI)          NB. the macro MB changes MELZI back into BOARD_MELZI
  #include "sanguino/pins_MELZI.h"
So the file is      sanguino/pins_MELZI.h
But each pins file can and often do include other files
eg #include "pins_SANGUINOLOLU_12.h" 
You have to see what each included file does, till you find the one you need. 
Yes sanguino/pins_SANGUINOLOLU_11.h has most things for your board.

"#define KILL_PIN 10"  should be all you need. If it is on D10

bill-orange, 2021-01-29:
@ellensp That was my thinking too.  Unfortunately "#define KILL_PIN 10" results in the following.

```
Marlin\src\MarlinCore.cpp: In function 'void manage_inactivity(bool)':
Marlin\src\MarlinCore.cpp:566:20: error: 'kill_state' was not declared in this scope
     if (kill_state())
                    ^
Marlin\src\MarlinCore.cpp: In function 'void minkill(bool)':
Marlin\src\MarlinCore.cpp:864:23: error: 'kill_state' was not declared in this scope
     while (kill_state()) watchdog_refresh();
                       ^
Marlin\src\MarlinCore.cpp:867:24: error: 'kill_state' was not declared in this scope
     while (!kill_state()) watchdog_refresh();
                        ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\sanguino1284p\src\src\MarlinCore.cpp.o] Error 1
```

I did a search and I had no better luck than VSCode in finding a kill_state function.

ellensp, 2021-01-29:
src/MarlinCore.h
```
#if HAS_KILL
  #ifndef KILL_PIN_STATE
    #define KILL_PIN_STATE LOW
  #endif
  inline bool kill_state() { return READ(KILL_PIN) == KILL_PIN_STATE; }
#endif
```
HAS_KILL is set when 
```
#if PIN_EXISTS(KILL)
  #define HAS_KILL 1
#endif
```

bill-orange, 2021-01-29:
@ellensp Hurrah, its working! For some reason `inline bool kill_state() { return READ(KILL_PIN) == KILL_PIN_STATE; }` was commented out.  After putting it back in, all was well.

Thanks!

I will close this.

closed 2021-01-29 by bill-orange

ellensp, 2021-01-29:
I've added a PR to add these missing pins to marlin going forward