MarlinFirmware/Marlin #20921: [BUG] Temperature no longer reported to OctoPrint

viper93458, 2021-01-29:
### Bug Description

Somewhere between yesterday morning and the most recent merges within 24 hours, OctoPrint can no longer read temperatures from my SKR Pro 1.0. It worked when I used Bugfix 2.0 from yesterday morning sometime (-8 PST) but when I used the latest bugfix this morning the issue cropped up. It causes OctoPrint to disconnect due to multiple timeouts.

### Configuration Files
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/5895797/Configs.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

Use the current bugfix 2.0 and connect to SKR 1.0 with Octoprint. It appears to connect OK but no temperature data is reported. This causes OctoPrint to throw an error about multiple timeout retires exceeded.

**Expected behavior:**

Temperatures would display

**Actual behavior:**

No temperature is displayed and the host disconnects.

#### Additional Information

I will do my best to provide any additional details as needed. I am not a developer so please be gentle. Thanks for your wonderful work and contributions to 3D printing and Open Source!

lightmaster, 2021-01-29:
I was able to roll back to the commit 9e004a94963771167bb3f5d6fa51ccd671a27a19 and it started working again, so it must be something related to PR #20913.

viper93458, 2021-01-29:
@lightmaster Thanks for the advise! I'll see if I can accomplish that. Unless a fix comes forth sooner. :)

TheLion, 2021-01-29:
I noticed the same thing on my SKR 1.3. I did notice that a temperature is reported to Octoprint when a print is started and Octoprint doesn't time out anymore.

When looking at the Terminal windows and even through Pronterface, no temperature is reported until a print is started.

GMagician, 2021-01-29:
I confirm. It seems autotemp report is not working

GMagician, 2021-01-30:
After digging into code it seems that is called the template function (empty) instead of overriden one

Z18Aplha, 2021-01-31:
Is there a quickfix?

GMagician, 2021-01-31:
Not yet, I think @thinkyhead should give a look at code to understand why override is not taking place

benlye, 2021-01-31:
> Is there a quickfix?

This worked for me:
https://github.com/MarlinFirmware/Marlin/pull/20913#issuecomment-770227265

GMagician, 2021-01-31:
yep that may solve but I think that virtual will add a cost in its call that is not desired

gimpfenlord, 2021-01-31:
> > Is there a quickfix?
> 
> This worked for me (one line change in autoreport.h):
> [#20913 (comment)](https://github.com/MarlinFirmware/Marlin/pull/20913#issuecomment-770227265)

can confirm this worked also for me.

Z18Aplha, 2021-01-31:
> > Is there a quickfix?
> 
> This worked for me (one line change in autoreport.h):
> [#20913 (comment)](https://github.com/MarlinFirmware/Marlin/pull/20913#issuecomment-770227265)

thx! i disabled the autoreport timeout in octoprint instead, works fine too. Octoprint polls Temperature now.

X-Ryl669, 2021-01-31:
The fix I've proposed is not optimal. We'll find a better solution.

X-Ryl669, 2021-01-31:
Can you try #20959, it should solve the building issue and also re-enable auto reporting correctly.

benlye, 2021-02-01:
> Can you try #20959, it should solve the building issue and also re-enable auto reporting correctly.

Working here.

viper93458, 2021-02-01:
Seems to be working for me again so I will close this out. Thanks!

-William

PR #20959 (2021-02-01): Fix AutoReporter not building and not working
### Description

Newly added AutoReporter class does not build correctly when MULTI_SERIAL is enabled. 
Also, it does not work since it the base class is not using a virtual auto_report method.
This PR fix both issue (I hope, I don't have any printer with those)
I'm not using a virtual table for the cost it has, but CRTP again, so it should be optimal.

Fix #20921 and #20913.

### Requirements

Board from #20921

### Configurations

Don't have

### Related Issues

#20921
closed 2021-02-01 by viper93458