MarlinFirmware/Marlin #20922: [BUG] BLtouch not compensating print

sbcu, 2021-01-29:
### Bug Description

The BLtouch starts the print by probing the bed correctly. But after I start a test print the z is not compensating for the warped bed.   

### Configuration Files

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/5895774/config.files.zip)

I am using a ender 3 with a Creality 4.2.7 board and a v3.1 bltouch

### Steps to Reproduce

1. I updated the firmware with the configuration files above.
2. I used the z probe offset wizard.  
3. then I ran the a test print.

**Expected behavior:**

I expected the probe to measure the bed. When I start a print, I expected, the test print would show the same results across the bed.

**Actual behavior:**

The probe measures the bed correctly. Although when the test print starts I can tell the printer is not compensating for the warping

What I have tried:

I have tried editing the gcode with a M420 S1 after G29.
I also tried editing the gcode with G29 then M500 then M420 S1

ellensp, 2021-01-29:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

sbcu, 2021-01-30:
[log.txt](https://github.com/MarlinFirmware/Marlin/files/5897103/log.txt)
I used octoprint terminal to create this log and I ran the commands above and the z probe offset wizard I also include the new configuration files.
[new_config.zip](https://github.com/MarlinFirmware/Marlin/files/5897105/new_config.zip)
I was already running bugfix-2.0.x downloaded today and there was no change.

jes00, 2021-02-01:
I'm also having this problem. I've tried 2.0.7.2, 2.0.7.1, and 2.0.1. But I used to not have the problem with 2.0.1 so I'm really at a loss at this point. It probes the bed and leveling is on, but it just doesn't compensate for it for some reason.

hamster65, 2021-02-01:
Do you see movement on the Z axis or is it not moving at all?

sbcu, 2021-02-01:
I have taken pictures of my test print to hopefully add some more info
[test print pictures.zip](https://github.com/MarlinFirmware/Marlin/files/5906424/test.print.pictures.zip)
as you can see on some of the squares the nozzle is too close and some the nozzle is farther away

> Do you see movement on the Z axis or is it not moving at all?

No I did not see any noticeable changes in the z axis

CRCinAU, 2021-02-02:
Try a 5x5 grid at a minimum.... a 3x3 grid when you have a measured difference of ~0.1mm will be altered all over the shop.

I'd also try increasing the PROBE_MARGIN to 20mm and see if your grid still looks anywhere near as bad....

jes00, 2021-02-04:
After further testing my Z axis IS moving, but I still have spots on my bed where the nozzle is too close to the bed.
Just got a raspberry pi and used the bed visualizer. I'm not sure how the results are so wonky, since I have a flat glass bed I'd expect the result to just be a tilted plane.

[image]

sbcu, 2021-02-06:
[image]

> Try a 5x5 grid at a minimum.... a 3x3 grid when you have a measured difference of ~0.1mm will be altered all over the shop.
> 
> I'd also try increasing the PROBE_MARGIN to 20mm and see if your grid still looks anywhere near as bad....

I tried this suggestion and the grid did get better but the Z is still not adjusting
The following are my thoughts on what is going on:
1. I think the issue is not the grid is inaccurate but that the printer is not using the grid data
2. I don't know if my Z is moving but if it is it is not doing enough

SeanReg, 2021-02-07:
>  but I still have spots on my bed where the nozzle is too close to the bed.

> I don't know if my Z is moving but if it is it is not doing enough

I seem to be having the same problems with mine. There was another issue regarding ABL that was closed without resolution in which I had added some comments a few weeks ago (#20105).

Here's what I notice, my prints are worse with ABL on then with ABL off. And often it actually seems like ABL moves the nozzle too close to the bad when it should bring it further. Almost like the Mesh is inverted. I'm currently resulting to turning off ABL so I can print without issues. I've even made my bed almost perfectly level, checked it with bed viz (looks level), then printed with ABL and have had worse results then with ABL off.  Something definitely seems broken here.

SKR Mini E3 V2
Ender 3 
Bltouch (3DTouch) 3.1

CRCinAU, 2021-02-07:
We're just spinning our wheels here... If nobody gathers the details from @ellensp's post above, nothing will ever get done from this.

The majority of these complaints are usually user error - so its up to you guys to convince us that there's actually a problem and it isn't user error. You do this by gathering all the info that's been requested.

Without that, just wave your hands around chanting "this isn't the bug you're looking for".

jes00, 2021-02-07:
I downloaded the latest bug fix version and configured it the same as my regular version and tried to build it but it just gets stuck on
> Processing STM32F103RC_btt_512K (platform: ststm32@~10.0; board: genericSTM32F103RC; framework: arduino)

techis, 2021-02-12:
HI All,

I am having a similar issue with a BL Touch on a SKR 1.3 on an Ender 3 Pro. I just rebuilt my config with the latest bugfix branch today. The config I have is in the config.zip. I have also logged the serial output from a few prints as well as running a G26 B60 H190 L0.2. That is attached as serial.txt. 

The first print is a 20 mm cube sliced with prusa slicer at 0.3. The second print is a 20 mm cube sliced with the same profile in prusa slicer at 0.2. These both printed with very little issue. The last print is a part cooling fan remix that doesnt stick on the skirt and the supports dont stick either. after the 3 prints I ran the G26 command which also looks like it is having some adheasion issues. 
[Config.zip](https://github.com/MarlinFirmware/Marlin/files/5969280/Config.zip)
[serial.txt](https://github.com/MarlinFirmware/Marlin/files/5969281/serial.txt)

If I can provide more information please let me know.

rechena, 2021-02-13:
I was just raised awareness for this open issue. 
I'm having similar issues, looking at my bed on Octoprint my bed is pretty leveled, and the Z Offset is based on a sheet of paper. I can see when a print starts that the probe is done and when printing things look a bit leveled, but then some parts of the bed still have adhesion issues. For example printing the ender 3 bed leveler test, I have some sides where the layer is squished against the bed and the other side is barely sticking. 

My settings:  
* Marlin-bugfix-2.0.x
* Ender 3 Pro with 4.2.7 silent board
* Oficial BLTouch. 

Let me know how can I provide more info.

jasonmacdonald, 2021-02-14:
I'm having the same issue with the newest build. From looking at the logs I never see any "Z" change as part of my "G1" calls for first layer, it's just pinned at 0.3 as my first layer height. I tried to grab the logs but it seems Octo-print is cutting them off so I'll have to find another way. I confirmed that my mesh is saved and can be recalled with M503. Started the print with "M420 S1 Z2" as well, but I don't see any z change logged nor does the Z axis move. Unless I'm completely misunderstanding how the mesh is applied shouldn't I see a "z" change as part of the "G1" command?

CRCinAU, 2021-02-14:
> Unless I'm completely misunderstanding how the mesh is applied shouldn't I see a "z" change as part of the "G1" command?

You're misunderstanding how the mesh is applied. It is done in firmware, on the controller - not via altering the G code sent to the printer.

jasonmacdonald, 2021-02-14:
> > Unless I'm completely misunderstanding how the mesh is applied shouldn't I see a "z" change as part of the "G1" command?
> 
> You're misunderstanding how the mesh is applied. It is done in firmware, on the controller - not via altering the G code sent to the printer.

Ah, my mistake then. How could you validate that the mesh is being applied properly?

sbcu, 2021-02-15:
> Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate: [...]

[log.txt](https://github.com/MarlinFirmware/Marlin/files/5983950/log.txt)
I just want to confirm that I did respond, this is the same file as in the 3 post on this issue
I appreciate the support on this issue

jasonmacdonald, 2021-02-16:
Ok, more strangeness and I'm not sure if it's related. I ran `M48` multiple times in the back right corner, never changing anything or even touching the printer.

First-round it reported:
```
Mean: 0.409917 Min: 0.356 Max: 0.463 Range: 0.107
Standard Deviation: 0.034312
```

By the 6th round it reported:
```
Mean: 1.281750 Min: 1.231 Max: 1.335 Range: 0.104
Standard Deviation: 0.033322
```

A change of almost a full millimeter to the Mean and yet the standard deviation between rounds stayed more or less the same.

Here's the full log.
```
Recv: Mean: 0.409917 Min: 0.356 Max: 0.463 Range: 0.107
Recv: Standard Deviation: 0.034312
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.546750 Min: 0.493 Max: 0.602 Range: 0.109
Recv: Standard Deviation: 0.034642
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.682833 Min: 0.628 Max: 0.738 Range: 0.110
Recv: Standard Deviation: 0.034267
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.820167 Min: 0.219 Max: 0.934 Range: 0.715
Recv: Standard Deviation: 0.202422
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 0.954500 Min: 0.353 Max: 1.067 Range: 0.714
Recv: Standard Deviation: 0.202299
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
Send: M48
Recv: M48 Z-Probe Repeatability Test
[...]
Recv: Finished!
Recv: Mean: 1.281750 Min: 1.231 Max: 1.335 Range: 0.104
Recv: Standard Deviation: 0.033322
Recv: 
Recv: X:169.00 Y:186.00 Z:10.00 E:0.00 Count X:13920 Y:15680 Z:4000
Recv: ok
[...]
```

I have no idea what could cause this. The probe is just moving up and down. Are my z-steps off or something? Could that be causing the print to be off?

dceddia, 2021-02-17:
I'm having a similar problem on my FolgerTech i3 2020. (posted about it on reddit as well [here](https://old.reddit.com/r/FixMyPrint/comments/ldh3ik/bltouch_bed_is_out_of_level_but_manual_bed_level/)).

I'll run the diagnostics mentioned above and report back.

In the meantime, has anyone tried the manual mesh leveling, with the `MESH_BED_LEVELING` option and a slip of paper?

On my printer, a manually-created 3x3 mesh seems near perfect ([image of first layer](https://preview.redd.it/zuaennbhypf61.jpg?width=1200&format=pjpg&auto=webp&s=f6ea3ed7e9547480b358cadae90d9458c6c122c8)), while a 3x3 or even 6x6 mesh done by the BLTouch and `AUTO_BED_LEVELING_BILINEAR` is consistently slanted ([this image](https://preview.redd.it/m3v3lpbhypf61.jpg?width=1200&format=pjpg&auto=webp&s=75944bddae141d545f40f5363c3b2850a55742f4)). This has got me very confused, but does seem to rule out mechanical issues like the bed being too far out of level to compensate, or the z axis being misaligned. It seems like it has to be something with the firmware or the BLTouch, unless there's something else I'm missing.

ManuelMcLure, 2021-02-17:
> On my printer, a manually-created 3x3 mesh seems near perfect ([image of first [...]

With a dual linear rod design like your FolgerTech it's very easy for the X axis to be twisted unless the rods are exactly parallel. Unless your Y probe offset is 0, this has the effect of causing the Z probe offset to vary depending on the X coordinate, resulting in a mesh that is higher on one side than the other. This is a mechanical issue. Even 0.5mm difference between the positions of the rods from side to side can result in a noticeable tilt in the mesh. I had similar issues with my Anet A8 - once I replaced the linear rods with a linear rail my meshes suddenly became perfect.

Generating a manual mesh uses the actual nozzle to measure the position, so it doesn't matter that the nozzle is rotating around the Z axis as it's corrected for.

dceddia, 2021-02-17:
> Even 0.5mm difference between the positions of the rods from side to side can result in a noticeable tilt in the mesh.

Thanks for this suggestion, I'll see what I can do about measuring that! Makes a lot of sense that the nozzle itself would be fine in that case.

I remeasured my probe Y offset as part of the troubleshooting and it was 1.1mm, so I might also try printing a new mount to get that centered too.

ManuelMcLure, 2021-02-17:
With a 1.1mm probe offset the effect of the twist will not be very big. The X offset also affects things but to a much lesser extent. So this may not be due to a twisted X axis.

hamster65, 2021-02-18:
I have seen this problem with levelling on my printer for some time. I think I have ruled out mechanical issues. There were a few issues on github but all were closed because it could not be reproduced or mechanical issues were found. I switched from bilinear to linear levelling and had far better results with that but still it seems off.

I have yet to try manual mesh levelling. From what I read in the other issues no one has tried that, I think.

Dceddias finding points to a problem while probing. Could this have something to do with the endstop noise threshold or endstop interrupt feature? Both are commented out on my setup but I wonder if there is some "averaging" or interrupt blocking going on in the code that would throw off probing. It's just a wild guess, though.

tayloss, 2021-02-18:
I have come across this bug as I was going mad thinking there is something wrong with my machine! I have levelled the bed and used the Z probe offset to ensure its correct, but cannot print the test pattern on my Ender 5 Pro, Creality v4.2.7 board for the life of me! Changing the build surface to a PEI sheet made no change.. I have used octoprint bed visualiser to ensure the bed is level and its fine. 

During the print, I baby stepping to ensure the brim prints correctly and then as it started to print the main test pattern, the front is too close and the back of the bed prints in the air! One of the things i did to see if the Z is moving was to introduce a large dip in the bed and marked the z stew connector to see if it moved at all.. I think it did, but maybe not enough as the print was extruded in mid air and not on the build plate.. I am completely lost as my bed is warped in the middle, so if I do get the middle level, the edges are too close hence the reason for the BL touch.

For info, I have tried manual mesh bed levelling and had similar results, so am sure the bed levelling bit of the f/w is broken. I followed a few youtube video who recommended using the bugfix branch as the latest release. The reason i installed the BLT was to get over the Z stop end switch preventing me from levelling the dips. I am almost inclined to use it as an end stop only and forget the levelling function.

Has anyone tried UBL to see if it makes any difference?

I too have ripped the machine apart, greased everything, check flow, check e-steps and change PLA, nothing sticks to the bed now with the BLT enabled!

tayloss, 2021-02-18:
Just to add, has anyone enabled G26 to see if the test pattern works? I see it seems to need to be enabled in the configuration.h, and you can issue a G26 command for the pattern to print?

tayloss, 2021-02-18:
[image]

BL Touch auto bed levelling
[image]
Manually levelling bed

jasonmacdonald, 2021-02-18:
@tayloss Out of curiosity, are you using the full 5 pin connector on the 4.2.7 board or the z-stop pin? I'm using the full 5 pins on the BLTouch input, but I'm going to try switching to the z-stop once my JST connectors arrive tomorrow. Maybe that will make a difference?

tayloss, 2021-02-18:
> @tayloss Out of curiosity, are you using the full 5 pin connector on the 4.2.7 board or the z-stop pin? I'm using the full 5 pins on the BLTouch input, but I'm going to try switching to the z-stop once my JST connectors arrive tomorrow. Maybe that will make a difference? 

Yes, it's the full 5 pin connector as it was the Creality kit with the correct cable supplied.. I wonder if the pins are mapped incorrect for some reason, but the grid is clearly being created and saved as I can see it in octoprint and when entering M503.

There is something not right, as I can almost print with manually levelling, but I have a dip in the middle I cannot seem to get rid of even with foil..

dceddia, 2021-02-18:
> I have a dip in the middle I cannot seem to get rid of even with foil

@tayloss Could you (or have you already) tried using more probing spots with the manual mesh leveling? I think this is the line to look for in Configuration.h:

`#define GRID_MAX_POINTS_X 3    // Don't use more than 7 points per axis, implementation limited.`

It's inside the `#elif ENABLED(MESH_BED_LEVELING)` block, around line 1339 for me.

Maybe change that number to 7, spend forever with a piece of paper for those 49 points, and see how it does? 😄

hamster65, 2021-02-18:
Number of points makes no difference for me, I already tried that (3x3 and 7x7).

rechena, 2021-02-18:
> @tayloss Out of curiosity, are you using the full 5 pin connector on the 4.2.7 board or the z-stop pin? I'm using the full 5 pins on the BLTouch input, but I'm going to try switching to the z-stop once my JST connectors arrive tomorrow. Maybe that will make a difference?

I'm using the 2 pins for power on the Z and the 3 other pins on the 5 pin connector and have the same problem :(

tayloss, 2021-02-18:
Same here.. manual mesh bed levelling "seems" to have the same bug and is linked.. I know the middle is level and the points around the dip and I still get the print extruding mid-air... 

I have lost faith in the bed levelling as of now... other people are reporting successful prints, but using the Creality download, so may try that to see if it's working correctly...

Has anyone tested the main branch to see if the problem exists?

rechena, 2021-02-18:
> I have lost faith in the bed levelling as of now... other people are reporting successful prints, but using the Creality download, so may try that to see if it's working correctly...

If you do this test do you mind letting me know? The reason I compiled Marlin was because the Creality one didnt came with the M600 enabled, but I rather have proper bed leveling.. Thanks

jasonmacdonald, 2021-02-19:
> Happy to help if anyone wants to try UBL as well to test their code too?

I haven't looked deeply at UBL, is there a large difference between that and using the BLTouch settings?

rechena, 2021-02-20:
> Happy to help if anyone wants to try UBL as well to test their code too?

What's the big difference in terms of configuration with the bltouch? Would love to try to see if I can get a decent print. Thanks

tayloss, 2021-02-20:
I used this guide and it worked fine first time.. you still babystep the first layer, but if the print still doesn't still you can adjust individual probe points.

 https://www.3dmakerengineering.com/blogs/3d-printing/unified-bed-leveling-marlin

Also, make sure when you use G29 P1 you follow it up a couple of times with G29 P3 which fills in the bits of the bed it cannot probe..

This will confirm if this but if the bed levelling code is working.

I have successfully printed a gear cube for the first time!

rechena, 2021-02-20:
Seems easy enough, would still like to know if there's really an issues with the mesh one or it's just me :(

rechena, 2021-02-20:
> I used this guide and it worked fine first time.. you still babystep the first layer, but if the print still doesn't stick you can adjust individual probe points.
> 
> https://www.3dmakerengineering.com/blogs/3d-printing/unified-bed-leveling-marlin

How, just followed this guide and used the option` #define GRID_MAX_POINTS_X 10` this is gonna take a while... 100 probes... might reduce to something smaller for sure.

tayloss, 2021-02-20:
It doesn't do all 100 as there is bits of the bed it can't reach, so use G29 P3 to fill in the gaps.. do you use Octoprint? 

If so, look at the grid and you will see some of the edges aren't completely filled in as the probe can't measure.. so the command above fills it in and then you can edit the points to make manual adjustments.. takes a while, but seems this is the only method that works for me as reverted to Bilinear and saw the same issue as you..

Not sure what else to report on this bug as it could be board specific?

rechena, 2021-02-20:
> Not sure what else to report on this bug as it could be board specific?

I might be wrong but I've seen people with other boards like the mkr reporting similar issues :( 
Humm still its probing 57 times which takes forever, is there a way to not have to do this everytime I start a print? 

:/

tayloss, 2021-02-20:
Yes.. in you starting G code, you add G29 L0 if you saved your grid in slot 0 or L1 for slot 1 then G29 J which then does a 3 point check at the start of the print..

Make sure you enable it by a G29 A, M500 then do a M503 and check that it's says active grid slot =x where you have save it! 

To prove, use G29 T to view the grid..

It took a while on my 5 pro, but it has resolved the BLT issue I had the same as you!

I have also made the bed very unlevel and can feel it correcting more so than the other methods.

rechena, 2021-02-20:
Ok, I've did all the steps in the tutorial using the commands:

```
M190 S{material_bed_temperature} 
G28 
G29 P1 
G29 P3 
G29 F10 
G29 S1 
G29 A 
G29 L1 
M500
```
I've then added to the slicer settings the following at the start of the gcode:

```
; Trying UBL 
G28
G29 L1 
G29 J 
G29 F10 
G29 A
M500
; UBL END
```

And will try a print now, but my question is, should I then every now and then run the full probe to get a new updated mesh?

Oh, one note, for anyone following the tutorial on that link, dont forget to configure the Z offset again, cause I did :P 

Thanks for the help.

rechena, 2021-02-20:
Might be this bed test but although it looks better I still see some places where the filament is squished and others not... might be the z offset though... [image]

tayloss, 2021-02-20:
I don't think you need to do G29 in the start Gcode as it's active if you do G29, M500..

If you have a place that's still not right, go into the menu and edit the grid.. it's a graphics display that you can turn the wheel and move to the point you need to adjust... then it checks the points around it as well...

rechena, 2021-02-20:
> I don't think you need to do G29 in the start Gcode as it's active if you do G29, M500..

So just

```
; Trying UBL 
G28
G29 J 
G29 F10 
G29 A
M500
; UBL END
```
> If you have a place that's still not right, go into the menu and edit the grid.. it's a graphics display that you can turn the wheel and move to the point you need to adjust... then it checks the points around it as well...

Oh, will check it now. Thanks :)

tayloss, 2021-02-20:
Sorry.. G29 A is what I meant!

G29 L1
G29 J 

Is all I have in mine and it works! Well, the Z moves... 

I have retested bilinear and I'm back to printing mid air and too close at the back... so I think that Bilinear is broken or my bed is so warped that it can't compensate....

rechena, 2021-02-20:
It's still a nope for me :( so it must be me... [image]

hamster65, 2021-02-20:
I don't see a difference in bed visualizer between bilinear and UBL. But the first layers of the two small prints I did with UBL looked great! It's too early for conclusions, though.

tayloss, 2021-02-20:
> I don't see a difference in bed visualizer between bilinear and UBL. But the first layers of the two small prints I did with UBL looked great! It's too early for conclusions, though.

That's the point, the bed hasn't changed between a 7x7 probe and the UBL, but I can clearly see the Z screw moving the bed with it not so much with bilinear.. it's very odd and I don't have the answer... I'm done with playing with the firmware and have concluded that UBL is worth the extra time to setup.. the manual probe point editing is a no brainer and works well as my test print is almost perfect all over and certainly couldn't achieve that with bL levelling...

On a side note, I see the other reported PR has been closed.. not sure what they are going to do about this one as I'm convinced it's broken...

hamster65, 2021-02-20:
I did not do any manual editing. Just
G29 P1 T  ;Probe
G29 P3 ;Auto Fill missing, repeat this several times
G29 A ;enable UBL
M500 ;save

This probed 100 points and since all were reachable, I don't think I have to probe additional points manually.
I am not sure if (and how often) G29 P3 is needed.

And the start Gcode (the config is set to enable levelling after G28):
G29 L1
G29 J

Report for Bed visualizer:
G29 T

tayloss, 2021-02-20:
So to add and not sure if this is an input to the issue...  but reports are that v4.2.2 boards are working fine with main branch and bugfix! 

Seems that v4.2.7 boards have the issue with main and bugfix... how do we prove this and submit logs to see what's going on? I know some SKR boards have also reported this issue.. so am beginning to believe it now!

rechena, 2021-02-20:
I hope so, cause I"m starting to loose my hope :( Just printed something. After that tried to print and on the gcode I had:

```
G28
G29 J 
G29 F10 
```

And on the second print its back miles away from the bed on one side :( I did nothing in between prints except remove the print... :(.

ManuelMcLure, 2021-02-20:
> I prefer(red) bilinear because it extrapolates beyond the grid which UBL does not. I wonder if this could be added? What happens now for my outer 1 cm of the bed where I cannot probe because of the clamps? Does is stay at the same height or does it jump to a Z-compensation of zero? The documentation does not explain this...

It's just a difference in philosophy. With Bilinear, the mesh is limited to the area that can be probed so it needs to extrapolate outside that mesh. With UBL, you define the mesh as covering the whole bed (i.e. including those 10mm near the edge that you cannot probe). Those points are not filled in by the `G29 P1`, but then you can run `G29 P3` and UBL will use extrapolation to fill those points so you have a complete mesh. The advantage of UBL is that you can then adjust those points if needed to get better results near the edge of the bed, while with Bilinear you are stuck with whatever the extrapolation is.

jasonmacdonald, 2021-02-21:
Well, I've now spent days trying different settings, using bi-linear and UBL and nothing seems to work. I went back to just wiping everything, turning off the BLTouch, and doing manual bed leveling... prints are now perfect!

So, either my BLTouch is broken, or something is seriously wrong with the bed leveling code. I give up trying to figure out what. I've spent a week and a half total fiddling with this. I'll accept the slightly less accurate manual leveling with a piece of paper.

techis, 2021-02-21:
I reflashed my SKR 1.3 with Cheetah from Kay3D that worked with the BL Touch. My first layer still needs some tuning but it is able to lay down a good layer. I would be happy to post the files if that helps. Maybe there was a change to how the Bilinear leveling happened in Cheetah versus the bugfix branch?

rechena, 2021-02-21:
> I reflashed my SKR 1.3 with Cheetah from Kay3D that worked with the BL Touch. My first layer still needs some tuning but it is able to lay down a good layer. I would be happy to post the files if [...]

That might be helpful indeed what is really wrecking my brain is what this "bug is not being a acknowledged :( so many people reporting and yet there's 0 reply.

CRCinAU, 2021-02-21:
> That might be helpful indeed what is really wrecking my brain is what this "bug is not being a acknowledged :( so many people reporting and yet there's 0 reply.

That's because in 56 comments, there's just about no useful information at all. Like every one of these "OMG, muh BLTouch don't work" threads, it turns into a dogpile of useless and unrelated information - of which, zero can be actioned.

As such, this should probably be closed like the rest of them.

rechena, 2021-02-21:
> That's because in 56 comments, there's just about no useful information at all. Like every one of these "OMG, muh BLTouch don't work" threads, it turns into a dogpile of useless and unrelated information - of which, zero can be actioned.
> 

Thats absolutely fair, altough I find your answer  a bit rude, there has been people providing information, yes there has been a lot of "useless" comments, but we though we were adding to the discussion. I suppose I need to apologize for that. I recall asking and other also, what more information can we provide and we got 0 replies. Again if I upset you I apologize, but that does not give you the right to be rude. For me you can close this, I won't be posting anymore.

techis, 2021-02-21:
@rechena, well said. if you are interested in still troubleshooting the issue, I would be happy to work with you. I forked marlin into my github, maybe we can continue our troubleshooting there?

CRCinAU, 2021-02-21:
> altough I find your answer a bit rude

That's fine - but you have to realise that this isn't the first, and it won't be the last thread. Just about zero of which have had any real credible information or diagnosis. People think every issue from filament problems, twisted chassis, dirty beds, filament adhesion issues etc etc etc are all because of the BLTouch - so therefore it must not be working.

If there is a bug, great - then people need to do the diagnostic steps that were given in this thread - and it'd need multiple people showing the same kind of results. If there's no correlation, then its highly unlikely to be a common bug or implementation issue related to leveling. Yet there's so many factors that nearly all are different - which leads to the inevitable close of the report.

It's not up to developers to go through 50+ posts and try to figure out what is common between them, what is not related at all, and what is likely user error. There just isn't enough time in the day to do that.

[36 more replies]

closed 2021-04-15 by sbcu