MarlinFirmware/Marlin #20926: Rapid stop restart without losing steps

mron, 2021-01-30:
### Description

Make M0/M1 stop immediately without losing steps. I know this has been brought up before. I'm going to give it another go if it hasn't been resolved.

Respond properly to GRBL Commands
Stop/Restart is needed for CNC and Lasers

### Feature Workflow

I know there is a project:
Extra realtime reporting #19330
Has it been completed? AFAIK it loses steps when stopped.

thinkyhead, 2021-02-28:
In order to do this, all steppers must be slowed down to zero over time. The faster the current speed, the longer the time must be. The trick to accomplish this would be to add a duration multiplier that affects the timing of calls to `Stepper::ISR`. This trick will add computational overhead to all stepping, if only to check for the multiplier being 1 and branching, but it is the most straightforward approach.

mron, 2021-02-28:
I believe I had the slowing down and stopping working. I set the decelerate_after to the current count. If the end of the block step is reached before stopping, a new block becomes active, I set it to decelerate immediately. This seems to be working. Whats driving me nuts right now is the planner. When velocity reaches zero, I stop the stepper ISR, I save the block_buffer and the planner states, I save the queue. I clear the block_buffer and the queue. There should be no more moves. As soon as I restart the stepper ISR a new move is added to the block_buffer. I don't know where it's coming from. The SD reader is paused, the queue is empty.
I think it's a processed move waiting for a block_buffer slot, but that's just a guess.

mron, 2021-02-28:
I've tested the stopping logic. It seems to work over multiple segments.

mron, 2021-03-06:
I think I got it working! It’s only motion for now. I’m going to test it for a while. If I don’t find problems, how do I put it out for more testing?

mron, 2021-03-11:
Here's my fork:
https://github.com/mron/Marlin/tree/bugfix2.0.xautoreport
I need to add a way to abandon the feedhold.

thisiskeithb, 2021-09-24:
Added in https://github.com/MarlinFirmware/Marlin/pull/17462

closed 2021-09-24 by thisiskeithb