MarlinFirmware/Marlin #20933: [BUG] Marlin fails to boot when enabling UART3 on SKR 1.4

zeleps, 2021-01-30:
### Bug Description

On an SKR 1.4, when setting either serial port to 3 (UART3), the printer does not start. It appears that the firmware fails to initialize, and it restarts every 2 seconds, failing to reach boot screen. Unfortunately, I have no other means to debug the issue at this point.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5898646/Marlin.zip)

#### Additional Information

Tested in latest bugfix. UART3 pins (4_28 & 4_29) are unused. Tried a lot of different configuration options to no avail. Since the printer does not boot, I have no feedback on what's causing the issue, and I can't think of a way to debug this. Any debugging tips are welcome.

rhapsodyv, 2021-01-30:
Probably related to: #20783.
@X-Ryl669 can you take a look?

zeleps, 2021-01-30:
Interesting, I noticed the templates when looking into my issue, didn't know it was so recent. I'll try to roll back before #20783 and see if the problem persists. Cheers!

zeleps, 2021-01-30:
Indeed, I had a branch locally from Jan 9, no issue there. It looks like #20783 still has some bugs...

X-Ryl669, 2021-01-30:
Can you enable  #20492 code to get a backtrace ?

GMagician, 2021-01-30:
I found some reboot causes and fixed with #20938, you may give it a try, maybe can also fix your issue

zeleps, 2021-01-30:
@X-Ryl669 it seems the issue was introduced before your code was merged. @GMagician I'm trying to find the exact commit that caused the issue, after that I'll try your solution. 

I'll get back to you with conclusions within the next hour.

zeleps, 2021-01-30:
Ok, the commit that breaks my configuration is #20810. It's not an obvious issue and it's a bit late to work it out rn, I'll give it a try tomorrow.

GMagician, 2021-01-31:
> 
> 
> Ok, the commit that breaks my configuration is #20810. It's not an obvious issue and it's a bit late to work it out rn, I'll give it a try tomorrow.

Maybe you should only properly set KILL_PIN_STATE

zeleps, 2021-01-31:
I don't have a KILL_PIN.

The problem is that I have DEV_MODE enabled and setup() tries to write debug output to the serial before it's initialized. Moving serial port initialization at the top of the function solved the issue.

@Rockman18 please check if initializing serial ports before kill_pin etc works for you, cause I don't want to break your changes.

zeleps, 2021-01-31:
I have created PR #20944 to address the issue, if there's more discussion on this, it should continue there. @Rockman18 please verify that #20944 is not breaking your change.

closed 2021-01-31 by zeleps

Rockman18, 2021-01-31:
@zeleps : Check ! It's ok for me !