MarlinFirmware/Marlin #20947: [BUG] MKS WIFI module updating at each startup

Rockman18, 2021-01-31:
### Bug Description

On last bugfix-2.0.x
Update the MKS wifi module doesn't rename the "Mks_Wifi.bin" file to "MKS_WIFI.CUR" on robin nano v2.0. So it tries to reflash the module at each startup.
I don't know if it's a error during the updating process and the module is not updated or if it's only the renaming process that fails.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5899806/Configuration.zip)

### Steps to Reproduce

1. Put Mks_Wifi.bin firmware on TF Card and insert it in the printer
2. Start the printer. The wifi module flashing process starts. Then the main menu appears.
3. Restart the printer
4. The wifi module flashing process restarts.

**Expected behavior:**

1. "Mks_Wifi.bin" file is renamed in "MKS_WIFI.CUR"
2. A kind of "popup" message indicates if flashing succeed or failed 

**Actual behavior:**

1. "Mks_Wifi.bin" file is not renamed in "MKS_WIFI.CUR"
2. No message indicates if flashing succeed or failed 

#### Additional Information

ellensp, 2021-02-01:
According to void mks_wifi_firmware_update() in src/lcd/extui/lib/mks_ui/wifi_module.cpp
It certialy should rename the file
file.rename(curDir, (char *)ESP_FIRMWARE_FILE_RENAME);
where #define ESP_FIRMWARE_FILE_RENAME  "MKSWIFI.CUR"

Rockman18, 2021-02-01:
Yeah it should but it doesn't. Perhaps a problem with the flashing process. I don't know

closed 2021-03-11 by github-actions[bot]