MarlinFirmware/Marlin #20948: [FR] G code to inspect print progress

logiclrd, 2021-01-31:
### Description

Maybe this already exists, but I haven't seen it. Basically, when running an SD card print, there's a little progress bar on the LCD12864 screen that tracks how far through the G code it has gotten. For some unknown reason, presently, mid-way through quite a long print, my LCD12864 screen has gone all wonky (maybe needs a reset? maybe needs the cable reseated? maybe is dying? unsure, but it's not relevant to this feature request :-) ). As a result, it doesn't appear to be effectively updating the print times and progress information:

[image]

(I also can't use the menu, which is slightly annoying.)

It does not appear to have impacted the print itself, though, which is continuing on.

I can issue an `M31` to get the current print time, but what can I issue to find out the print _progress_?

As a work-around, I have been using `M114` to find the Z value, which is a decent approximation of progress as I can infer which layer it is working on.

### Feature Workflow

1. Start an SD card print.
2. Issue a particular G code command.
3. Get back progress information equivalent to seeing the progress bar and time elapsed & estimated on the LCD screen.

Roxy-3D, 2021-02-03:
Obviously there is a problem.   But it could be just about anything causing the problem.   (It could even be electrical noise on your LCD cables.)   We have seen a lot of LCD flakiness in the past year.   I wonder if adding a "LCD Reset and Update to current status" would help us diagnose this better.   It would be good to see your screen 'correctly'.

EvilGremlin, 2023-07-19:
Done in https://github.com/MarlinFirmware/Marlin/pull/24767

closed 2023-07-19 by thisiskeithb