MarlinFirmware/Marlin #20949: example config for skr mini e3 not compiling

Karabenta, 2021-01-31:
### Bug Description

compilation failing when trying to load:

.platformio\lib\WiFi101\src\WiFiMDNSResponder.cpp

I tried using the latest bugfix, then I replaced my custom config with the provided example for the skr mini e3 v1.2 but the error persists. I also tried example configs for v1.0 and v2.0, with the exact same result. 

### Configuration Files

As it I'm also unable to compile the "foolproof" provided example (also linked below) this doesn't matter. 

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/5899848/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/5899849/Configuration_adv.txt)

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2

### Steps to Reproduce

Just compile the provided example config for skr mini e3 v1.2 in marlin 2.0.7.1 or bugfix 2.0.x (31/01/2021).

**Expected behavior:**

An error-free compilation.

**Actual behavior:**

Compilation fails. 

#### Additional Information

The terminal output:
[compiling error.txt](https://github.com/MarlinFirmware/Marlin/files/5899851/compiling.error.txt)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

qwewer0, 2021-01-31:
For compiling bugfix-2.0.x, use https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x
and for 2.0.7.2, use https://github.com/MarlinFirmware/Configurations/tree/release-2.0.7.2

Karabenta, 2021-01-31:
Thanks for your quick reply!

I tried them, even downloaded the bugfix again, so I have the very latest version. 
But the error persists :(

qwewer0, 2021-01-31:
I wasn't able to reproduce the issue. Downloaded https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%201.2 example files and it compiled without any error.

Karabenta, 2021-01-31:
Interesting, that's exactly what I'm using. 

I guess it's not a bug then, must be a local problem with my PC. I'm using windows 10, visual code and platformio are up to date. Do you have any suggestions?

qwewer0, 2021-01-31:
Try "Upgrade PlatformIO Core" and "Update All" inside platformio, but maybe a system restart will solve it.

Karabenta, 2021-01-31:
I already did that, but it again, just to make sure. No difference.

Karabenta, 2021-01-31:
Only the ststm32 platform was updated from 10.0.1 to 11.0 . While compiling it was still using the older one though, so I replaced 
platform =  ${common_stm32f1.platform} with
platform = ststm32@~11 
in platformio.ini

Alas, it also made no difference, although the newest platfrom was used during compilation.

sjasonsmith, 2021-01-31:
Try deleting the .platformio directory and let it rebuild itself.

If deleting it worries you, you can rename it instead.

Karabenta, 2021-01-31:
do you mean C:\Users\user\.platformio ?

Karabenta, 2021-01-31:
Btw, I love you guys for bothering and taking your time on a sunday afternoon!

sjasonsmith, 2021-01-31:
Yes. I would also delete the .pio directory inside your Marlin directory as well.

Assuming you are using VS Code you can delete these and they will be re-created when you relaunch VS Code.

Karabenta, 2021-01-31:
A.W.E.S.O.M.E

that did the trick!!

closed 2021-01-31 by Karabenta