MarlinFirmware/Marlin #20950: Filament runout triggered at beginning of print

Bogdan-Vasile, 2021-01-31:
Hello!

I've been strugling with the filament runout sensor installed on an SKR 1.4. The problem is that when any print is started, the filament runout is triggered, and I can't print anything.

The sensor is coming from an Anycubic i3 Mega S. Basically, it is an enstop switch with a led indicator. All three pins are connected correctly to the E0Det port. The M119 command is reporting the states correctly, both open and triggered.
The P1.26 is assigned correctly to the filament runout pin (checked in code and with M43).
I installed a 1u capacitor on the sensor between the signal and ground to prevent problems related to noise, and checked again with M119.
I also played with pullup and trigger states, but only LOW state gives a correct behavior.
I used Marlin version 2.0.7.2 and bugfix-2.0.8.
I tried starting the print from board sd and octoprint.
I enabled the filament runout debug feature and saw that when the print is started, one of the first printed messages on the octoprint console is Filament OUT. Then, the counter is decreased and when the remaining distance goes to 0, the nozzle parks and the filament change is initiated. On 2.0.7.2, the runout happens again after the first filament change. Didn't test this on bugfix version.

I think there is a problem with the filament runout state initialization, but I'm not that sure.

Any help would be appreciated.

GMagician, 2021-01-31:
What do you get with M119, triggered or open, when filament is present?

Bogdan-Vasile, 2021-01-31:
Ah, ok. I didn't know that. I get open when filament is present. I will change the logic states. Thank you so much for your help.

GMagician, 2021-01-31:
> 
> 
> Ah, ok. I didn't know that. I get open when filament is present. I will change the logic states. Thank you so much for your help.

Yep, this confused a lot of people because in cnf you set level to properly detect filament runout but M119 reports the opposite (filament present)

ellensp, 2021-02-01:
This looks to be sorted. Closing.

closed 2021-02-01 by ellensp