MarlinFirmware/Marlin #20952: [BUG] SKR-E3 DIP v1.1 / Marlin-bugfix-2.0.x - "No Media" with SD Card

ziomkris1, 2021-01-31:
### Bug Description
Hello Friends! New user here.

I can upload new firmware through inserting SD card into motherboard SD slot, but when I am trying to print, on LCD menu it says "No Media". However, when I'm connecting printer through USB cable with PC, I can initialize M21 command in pronterface and then I can see media in printer without problem.

Marlin-bugfix-2.0.x
SKR-E3 DIP v1.1 with TMC2208 UART
Stock Creality Ender 3 LCD screen

### Configuration Files
https://github.com/ziomkris1/Marlin-bugfix-2.0.x-Config.git

### Steps to Reproduce

Configuration.h:
#define SDSUPPORT

Configuration_adv.h:
#define SD_SPI_SPEED SPI_QUARTER_SPEED
#define SDCARD_CONNECTION ONBOARD
#define BOOTSCREEN_TIMEOUT 1500

1. Have insterted SD card in motherboard SD card slot
2. Switching On Printer
3. Going into LCD Main Menu
4. Scrolling down - No Media

**Expected behavior:**
I want to be able to print through SD card without connecting printer to PC via USB and making commands in pronterface. It should work when I am switching ON printer with SD card in motherboard but also when I am inserting card when printer is switched ON, just like in stock Ender 3. I was just refreshing SD card on menu and everything was ready.

**Actual behavior:**
Printer does not recognize SD card for printing, just for uploading firmware

When printer is switched ON without SD card in motherboard, on title screen it shows "SD Init Fail" message, then I am inserting it and it shows "Media Removed". There is just "No Media" message on the bottom of main menu. Nothing else.

When printer is switched ON WITH SD card in motherboard, on title screen it shows "Ender-3 E3 DIP Ready." But also "No Media" on the bottom of main menu. 

#### Additional Information
I am new into that topic, so I was really trying to find any help and tutorial that I could.

-Downloaded Marlin-bugfix-2.0.x from Marlin Firmware site
-Copied examples of all four main files and pasted them in Marlin folder from Configurations/config/examples/Creality/Ender-3/BigTreeTech SKR E3-DIP V1.1/
-changed some codes in config/config_adv file as many had problems, which include:
Configuration.h:
#define SDSUPPORT

Configuration_adv.h:
#define SD_SPI_SPEED SPI_QUARTER_SPEED
#define SDCARD_CONNECTION ONBOARD
#define BOOTSCREEN_TIMEOUT 1500

-formatted SD card into FAT32 using SDCardFormatter

Sorry if I've missed something, as I've said, I am new into this site and firmware editing. Of course I will provide any additional info that you need! Just tell me what exactly.

[Marlin-bugfix-2.0.x-Config.zip](https://github.com/MarlinFirmware/Marlin/files/5900016/Marlin-bugfix-2.0.x-Config.zip)

gutbrod, 2021-02-02:
Have you tried finding and enabling the following, With the Bugfix ~2.0.0.8 I had to enable this to use the onboard SD Card reader instead of the Attached SD Card reader via the BigTreeTech TFT 3.5 LCD E3 V3.
Around line 1396 in Configuration_adv.h
#define SDCARD_CONNECTION ONBOARD

ziomkris1, 2021-02-03:
Hi! 
#define SDCARD_CONNECTION ONBOARD I have already enabled, as I've told on my first message. I will try to use maybe other version of Marlin and see if something will change and will write here.

NezSez, 2021-02-04:
Might check these in Configuration_adv.h :
 
 // The standard SD detect circuit reads LOW when media is inserted and HIGH when empty.
  // Enable this option and set to HIGH if your SD cards are incorrectly detected.
  //#define SD_DETECT_STATE HIGH

  //#define SD_IGNORE_AT_STARTUP            // Don't mount the SD card when starting up

  // Enable if SD detect is rendered useless (e.g., by using an SD extender)
  //#define NO_SD_DETECT

ziomkris1, 2021-02-05:
I am very happy to announce that command given by NezSes:

//#define SD_DETECT_STATE HIGH
worked!

WAS: #define SD_DETECT_STATE HIGH
NOW IT IS: //#define SD_DETECT_STATE HIGH

SD card was reading new firmware, on "home menu" there was info that Ender 3 is ready, went into menu and on the bottom I could read files. Also, when I've removed card from the slot, there was "media removed" info on the screen, and when I've inserted card back, there was "Media inserted" info and without and problem I was able to find files on the bottom of the menu.

Thanks a lot! Simple solution that costed me almost whole week of headache

NezSez, 2021-02-05:
Excellent! All thanks goes to the Marlin devs.

rhapsodyv, 2021-02-26:
@ziomkris1 is everything ok and this can be closed?

ziomkris1, 2021-02-26:
Yes, I'm closing it. 

Sorry for long delay, I was doing a lot of testing to make sure everything is alright, since I didn't saw other topic where changing SD Detect State was a solution. I have another small problem but since it is not related I will make later a new thread.

Thank You again for help!

closed 2021-02-26 by ziomkris1