MarlinFirmware/Marlin #20953: [BUG] E1 steper motor not moving

aero4, 2021-01-31:
### Bug Description

I have a custom built 3D printer with Marlin 2.0.7.2 as firmware. I tried to set up my dual extruder but the problem is that the second extruder (E1) stepper motor is not moving at all. The hardware part works because when I swap the E0 and E1 extruder plug on my BIGTREETECH SKR V1.3 the E1 motor works without problems.
I also changed the pins file so that the Y STEP/DIR/ENABLE pins are the same as the one from E1 and the stepper motor also worked so it is not a problem concerning my TMC2130 motor drivers.

**I also tried to swap the pin numbers for E1 and E0 in the pins file and strangely the same motor (E1) didn't work again...**

I also tried Marlin 2.0.5.3 and the bugfix branch which did not help...

### Configuration Files

I also added the following lines to "pins/lpc1768/pins_BTT_SKR_V1_3.h":
#define E1_STEP_PIN                        P0_01
#define E1_DIR_PIN                         P0_00
#define E1_ENABLE_PIN                      P0_10
#ifndef E1_CS_PIN
  #define E1_CS_PIN                        P1_01
#endif
This also didn't solve the problem.

### Steps to Reproduce

1. download Marlin 2.0.7.2
2. add my configuration files
3. modify the pins file according to the lines stated above
4. upload to printer with dual extruder and BIGTREETECH SKR V1.3 board

**Expected behavior:**

Second stepper motor (E1) is working...

**Actual behavior:**

Second stepper motor does not work...

#### Question

* What can cause such a behavior?
* Is there something else which I can test?

ellensp, 2021-01-31:
no mention of how your testing
you do know you have to issue a T1 and that swaps E0 with E1?

ellensp, 2021-02-01:
You don't need to add these E1 defines 
the last line of pins_BTT_SKR_V1_3.h is #include "pins_BTT_SKR_common.h"
and in pins_BTT_SKR_common.h is
```
#ifndef E1_STEP_PIN
  #define E1_STEP_PIN                      P0_01
#endif
#ifndef E1_DIR_PIN
  #define E1_DIR_PIN                       P0_00
#endif
#ifndef E1_ENABLE_PIN
  #define E1_ENABLE_PIN                    P0_10
#endif
```

ellensp, 2021-02-01:
Cannot reproduce this issue using bugfix on a skr 1.3 with tmc2209's stepper drivers and FYSETC_MINI_12864_2_1 lcd (as that is what hardware I have)
G1 E20 moves E0 as expected
T1 changes tools as expected
G1 E30 move E1 as expected

aero4, 2021-02-01:
> 
> 
> no mention of how your testing
> you do know you have to issue a T1 and that swaps E0 with E1?

Thanks for your replies!

Yes I forgot to mention that. I used Simplify3D's control panel to change the tools. And there I saw that no cold extrusion warning occured.

> You don't need to add these E1 defines
> the last line of pins_BTT_SKR_V1_3.h is #include "pins_BTT_SKR_common.h"
> and in pins_BTT_SKR_common.h is
>
> ```
> #ifndef E1_STEP_PIN
>   #define [...]

Thanks for that info! I didn't see that, this makes sense...

> 
> 
> Cannot reproduce this issue using bugfix on a skr 1.3 with tmc2209's stepper drivers and FYSETC_MINI_12864_2_1 lcd (as that is what hardware I have)
> 
> after turning off cold hotend detection with M302 P1
> 
> G1 E20 moves E0 as expected
> T1 changes tools as expected
> G1 E30 move E1 as expected
> 
> Your config files updated for bugfix
> [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5901470/Configuration.zip)

Ok I will try the bugfix branch with the config files you provided!

aero4, 2021-02-01:
> Cannot reproduce this issue using bugfix on a skr 1.3 with tmc2209's stepper drivers and FYSETC_MINI_12864_2_1 lcd (as that is what hardware I have)
>
> after turning off cold hotend detection with [...]

I tried the commands.

T0 extruded without any problems

Then I switched to T1 and got the following error message:
```
SENT: G1 E30
READ: ok
SENT: M105
READ: ok T:22.50 /0.00 B:34.38 /0.00 T0:62.59 /0.00 T1:22.50 /0.00 @:0 B@:0 @0:0 @1:0
SENT: M105
READ: ok T:22.50 /0.00 B:34.22 /0.00 T0:62.22 /0.00 T1:22.50 /0.00 @:0 B@:0 @0:0 @1:0
READ: 
READ: E1 driver error detected: 0xC80C0000
READ: coil short circuit
READ: 		X	Y	Z	E	E1
READ: Enabled		false	false	false	true	true
READ: Set current	900	1100	1000	600	600
READ: RMS current	887	1049	994	581	581
READ: MAX current	1251	1479	1402	819	819
READ: Run current	28/31	18/31	17/31	18/31	18/31
READ: Hold current	14/31	9/31	8/31	9/31	9/31
READ: CS actual	14/31	9/31	8/31	9/31	12/31
READ: PWM scale	0	0	0	63	0
READ: vsense		1=.18	0=.325	0=.325	1=.18	1=.18
READ: stealthChop	false	false	true	true	true
READ: msteps		32	32	32	32	32
READ: interp		true	true	true	true	true
READ: tstep		max	max	max	max	max
READ: PWM thresh.					
READ: [mm/s]						
READ: OT prewarn	false	false	false	false	false
READ: triggered
READ:  OTP		false	false	false	false	false
READ: off time	3	3	3	3	3
READ: blank time	24	24	24	24	24
READ: hysteresis
READ:  -end		-1	-1	-1	-1	-1
READ:  -start		1	1	1	1	1
READ: Stallguard thrs	0	0	0	0	0
READ: uStep count	4	4	4	260	580
READ: DRVSTATUS	X	Y	Z	E	E1
READ: sg_result	0	0	0	0	0
READ: stallguard					
READ: fsactive					
READ: stst						
READ: olb						*
READ: ola						
READ: s2gb						
READ: s2ga						*
READ: otpw						
READ: ot						
READ: Driver registers:
READ: 		X	0x80:0E:00:00
READ: 		Y	0x80:09:00:00
READ: 		Z	0x80:08:00:00
READ: 		E	0x80:09:00:00
READ: 		E1	0xC8:0B:00:00
READ: 
READ: 
READ: Error:Printer halted. kill() called!
```
But even when I unplug the motor cable directly from my board this error occurs! What could cause this issue?

ellensp, 2021-02-01:
bad E1 stepper driver
Swap the stepper driver with E0 and try again.
If the error moves to e0, its defiantly a bad driver

aero4, 2021-02-01:
> 
> 
> bad E1 stepper driver
> Swap the stepper driver with E0 and try again.
> If the error moves to e0, its defiantly a bad driver

I tried that and swapped the driver two times. Always T1 causing this error...
This is for sure a Hardware issue I guess?

ellensp, 2021-02-01:
It does sound like its hardware.

You could check for shorts, remove the stepper driver and motor and check for shorts between the stepper motor pins

If you find one look for solder bridges on and around those pins. The traces from the stepper drivers and the stepper motor plugs is really short

aero4, 2021-02-01:
> 
> 
> It does sound like its hardware.
> 
> You could check for shorts, remove the stepper driver and motor and check for shorts between the stepper motor pins
> 
> If you find one look for solder bridges on and around those pins. The traces from the stepper drivers and the stepper motor plugs is really short

Ok, thanks for the tips! I will investgate that later today and will keep you updated!